realsenseai/librealsense #5011: Android Support for T265

tropicdragon, 2019-10-08:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Platform                                 | Android  |

### Issue Description

I noted that under librealsense/doc/android.md ....There is a commit that states: Add notes that T265 is not yet supported on android and macOS

I have successfully built on android, and plugged in a D415 and it worked. However, when I plug in a T265 it is not even detected. Is it still the case that T265 is not supported on android? Are there plans on it being supported in the future?

MartyG-RealSense, 2019-10-08:
Intel plan to add Android support for T265 but there is currently no time estimate for when it will be released.

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Hi @tropicdragon - Just keep monitoring our releases at this time we cannot comment on future release plans.  I will close this ticket out now.

Thanks @MartyG-RealSense for your comment here as well.

closed 2019-12-05 by RealSenseCustomerSupport

bfulkers-i, 2019-12-06:
@tropicdragon #5213 will lay the groundwork that will eventually lead to android support, hopefully sooner rather than later.