realsenseai/librealsense #5012: The frameset in rs2::frameset fs in intelrealsense C++ is equivalent to which object in pyrealsense?

Dontla, 2019-10-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
The frameset in rs2::frameset fs in intelrealsense C++ is equivalent to which object in pyrealsense?
The question is from [https://dev.intelrealsense.com/docs/rs-multicam](https://dev.intelrealsense.com/docs/rs-multicam).
[image]

lramati, 2019-10-28:
due to historical reasons, the same object in Python is still called `composite_frame`

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

@Dontla Any other questions for this? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@Dontla  Will close this issue if no other questions. Thanks!

closed 2019-11-22 by RealSenseCustomerSupport