realsenseai/librealsense #5013: What is the role of the as_frame() function in pyrealsense?

Dontla, 2019-10-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
What is the role of the as_frame() function in pyrealsense?
[image]

lramati, 2019-10-28:
`as_frame` is an artifact of how the C++ library is wrapped for python. It's job is to take a reference of any type that inherits from `frame` and return a reference to the same data that is of type `frame`.
Note that between Python's dynamic typing and the automatic casting capabilities of the wrapping layer, you should never need to call `as_frame()`

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @Dontla

Is there anything else needed on this thread?

closed 2020-02-07 by RealSenseCustomerSupport