realsenseai/librealsense #5016: depth image

tonyjonyjack, 2019-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I try to extract object contour from depth image,so I want to get  depth image like this (from realsense r300)
[image]

but realsense D435 provides depth image 
[image]
How can I get depth image like picture one by D435.Thanks in advance

agrunnet, 2019-10-09:
How close are you?  Looks like you are too close and need to move farther than 19cm away on left part of image.

tonyjonyjack, 2019-10-09:
> How close are you? Looks like you are too close and need to move farther than 19cm away on left part of image.

The image D435 provided is blurry ,I can't deal with it,I want to get to  depth image like this (from realsense r 300)
[image]
I am surprised that lateset product  D435 can't provide image like old product r300.

g2-bernotas, 2019-10-09:
@tonyjonyjack how close are you to the D435 camera? Have you tried different parameters provided in the RealSense Viewer?

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

@tonyjonyjack, please let us know if you need further assistance on this.
Thank you.

closed 2019-11-01 by RealSenseCustomerSupport