realsenseai/librealsense #5017: C# Wrapper - Won't Stop Recording - Cancellation Token inneffective

brenteickhoff, 2019-10-09:
|---------------------------------|------------------------------------------- |
| Camera Model | D400 series
| Firmware Version | 05.11.06.250
| Operating System & Version | Windows 10
| Platform | PC
| SDK Version | 2.19.1
| Language | C#
| Segment | Healthcare Research, AI Deep Learning, Motion Pattern Recognition

### Issue Description
Using C# Wrapper - Tutorial 2 - Capture, I have modified the code to Record to File.
When I call tokenSource.Cancel(); it stops streaming but continues recording to the file until the application shuts down. I placed all the code in a class and reference the window and image controls globally. Even if I call the cancel and then set the class to null, it keeps recording. I tried to wait 500 msec after Cancel and dispose the Pipeline and it errors because it is still recording.
I need to stream the data to the screen and start and stop recording. This is my code, it is the tutorial code with one line added (cfg.EnableRecordToFile(@"C:\test\ros.bag");

namespace ClassLibrary
{
    public class Camera
    {
        private Pipeline Pipeline;
        private Colorizer Colorizer;
        private CancellationTokenSource TokenSource = new CancellationTokenSource();

        public Camera()
        {
        }

        static Action<VideoFrame> UpdateImage(Image imageControl)
        {
            var wbmp = imageControl.Source as WriteableBitmap;

            return new Action<VideoFrame>(frame =>
            {
                using (frame)
                {
                    var rect = new Int32Rect(0, 0, frame.Width, frame.Height);
                    wbmp.WritePixels(rect, frame.Data, frame.Stride * frame.Height, frame.Stride);
                }
            });
        }

        public void Start()
        {
            try
            {
                Action<VideoFrame> updateDepth;
                Action<VideoFrame> updateColor;
                Colorizer = new Colorizer();
                Pipeline = new Pipeline();
                Config cfg = new Config();

                //**********MODIFIED CODE***********
                //cfg.EnableStream(Stream.Depth, 640, 480);
                //cfg.EnableStream(Stream.Color, Format.Rgb8);

                cfg.EnableStream(Intel.RealSense.Stream.Color, 424, 240, Format.Rgb8, 30);
                cfg.EnableStream(Intel.RealSense.Stream.Depth, 424, 240, Format.Z16, 30);
                cfg.EnableRecordToFile(@"C:\test\ros.bag");
                //**********END MODIFIED CODE***********

                PipelineProfile pp = Pipeline.Start(cfg);
                SetupWindow(pp, out updateDepth, out updateColor);

                Task.Factory.StartNew(() =>
                {
                    while (!TokenSource.Token.IsCancellationRequested)
                    {
                        using (var frames = Pipeline.WaitForFrames())
                        {
                            var colorFrame = frames.ColorFrame.DisposeWith(frames);
                            var depthFrame = frames.DepthFrame.DisposeWith(frames);
                            var colorizedDepth = Colorizer.Process(depthFrame).As<VideoFrame>().DisposeWith(frames);
                            Global.DaqWindow.Dispatcher.Invoke(DispatcherPriority.Render, updateDepth, colorizedDepth);
                            Global.DaqWindow.Dispatcher.Invoke(DispatcherPriority.Render, updateColor, colorFrame);
                        }
                    }
                }, TokenSource.Token);
                Thread.Sleep(500);
                //pp.Dispose();
                //cfg.Dispose();
                //Pipeline.Dispose();

            }
            catch (Exception err)
            {
                throw err;
                //MessageBox.Show(ex.Message);
                //Application.Current.Shutdown();
            }
        }

        public void Stop()
        {
            TokenSource.Cancel();
        }

        private void SetupWindow(PipelineProfile pipelineProfile, out Action<VideoFrame> depth, out Action<VideoFrame> color)
        {
            using (var p = pipelineProfile.GetStream(Intel.RealSense.Stream.Depth).As<VideoStreamProfile>())
               Global.ImageDepthControl.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            depth = UpdateImage(Global.ImageDepthControl);

            using (var p = pipelineProfile.GetStream(Intel.RealSense.Stream.Color).As<VideoStreamProfile>())
                Global.ImageColorControl.Source = new WriteableBitmap(p.Width, p.Height, 96d, 96d, PixelFormats.Rgb24, null);
            color = UpdateImage(Global.ImageColorControl);
        }
    }
}

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

Hi @brenteickhoff,

Could you please provide us with the error messages you see indicating that recording is still on while the streaming is off.

Thank you!

RealSenseCustomerSupport, 2019-12-26:
----------------------------------------------

Hi @brenteickhoff,

Do you still need help with this question?

closed 2020-01-07 by RealSenseCustomerSupport