realsenseai/librealsense #5018: D435i: The actual FPS of gyro stream too high

terranas, 2019-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435I | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                            | 2.19.0, 2.29.0 | 
| Language                            | C++ | 

### Issue Description
Even though I set to 200Hz as gyro stream FPS, the actual FPS is 2000Hz.
I observed this at Viewer v2.29.0 and v2.19.0, with Firmware version 05.11.01.100.
It is no longer observed after updating Firmware Version to 05.11.15.0.
Which version has been fixed from?
I didn't know from the Specification Update document.

[image]

terranas, 2019-10-09:
Is this the same problem as below in #5003 ?

> D435i – IMU frame drops

ev-mp, 2019-10-10:
@terranas hello, the FW version v5.11.1.100 is indeed lower than the recommended.
The mentioned [D400 Specifications Update](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf) document depicts only the changes recorded in the publicly available firmware versions, but does not elaborate changes incorporated in the intermediate versions. So it might be the case that the fix was incorporated in between.

We will keep the ticket open for a while, to monitor the situation with v5.11.15.0. Please update should the issue reappear with that FW version.

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Hi @terranas,

Do you still find this issue with FW 5.11.15.0 per your recent use?

closed 2019-11-26 by dorodnic