realsenseai/librealsense #5020: Hardware-accelerated video decoding

Panepo, 2019-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.25.0                          | 
| Language                            |  C/C#/opencv/python                          | 
| Segment			|  others                           | 

Hi there,

I am wondering if it is possible to accelerate the speed while streaming by hardware-accelerated video decoding? Since I need more CPU computing power to process other stuff like deep learning.

I found that there is a method to make Media SDK as OpenCV video decode backend like [this](https://github.com/opencv/opencv/wiki/MediaSDK-encode-decode-backend). Is there a similiar method or librealsense already build-in hardware-accelerated video decoding?

Thanks

MartyG-RealSense, 2019-10-09:
I hope that the link below will be useful to you.

https://github.com/IntelRealSense/librealsense/issues/4905#issuecomment-533854888

Panepo, 2019-10-10:
@MartyG-RealSense Thanks a lot, it is helpful.

closed 2019-10-10 by Panepo