realsenseai/librealsense #5021: install pyopengl_accelerate

GOBish, 2019-10-09:
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | () | 
| Operating System & Version |  Raspian Buster |
| Kernel Version (Linux Only)    |  4.19.75 |                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  { 2}                          | 
| Language                            |  { }                          | 
| Segment			|  { }                          | 

### Issue Description
I am trying to use a realsense D435 with a Raspberry Pi 4 using the instructions here 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

along with some modifications to the CMakeLists.txt. It is working!

The only thing that did not work was this step:

$ sudo -H pip3 install pyopengl_accelerate

under "change pi settings (enable OpenGL)"

I get
Failed building wheel for pyopengl-accelerate

and then at the end
Command "/usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-rx4uc_ia/pyopengl-accelerate/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-aezekfy5/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-install-rx4uc_ia/pyopengl-accelerate/

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

@GOBish Sorry for late response. Did you get through the issue? Please try to get more support on python related forums.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@GOBish Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport