realsenseai/librealsense #5022: C# Wrapper - Can't Build Working Tutorials - Issue with DLL(s) in recent Releases

brenteickhoff, 2019-10-09:
### Required Info:
- Camera Model - D400 series
- Firmware Version - 05.11.06.250
- Operating System & Version - Windows 10 
- Platform – PC (64bit)
- IDE - Visual Studio 15 2017
- SDK Version - 2.19.1 (working), 2.24.0 & 2.29.0 (not working)
- Language - C# WPF
- Segment - Healthcare Research, AI Deep Learning, Motion Pattern Recognition

### Issue Description
I am able to successfully build and run working samples for C# wrappers in SDK v.2.19.1.
I cannot do the same for versions 2.24.0 or 2.29.0, there seems to be a problem with the DLLs

### Steps to Reproduce: WORKING Tutorial Project in v.2.19.1
Steps to Build and Run a successful functional C# Tutorial Project from Source Code v.2.19.1
1.	Download ‘librealsense-2.19.1.zip’ from [https://github.com/IntelRealSense/librealsense/releases]
2.	Unzip files to ‘C:’
3.	Create directory ‘build’ (C:\librealsense-2.19.1\build)
4.	Using CMake v.3.14.0 GUI 
   a.	Click ‘Browse Source…’ and browse to directory ‘C:/librealsense-2.19.1’
   b.	Click ‘Browse Build…’ and browse to directory ‘C:/librealsense-2.19.1/build’
   c.	Click ‘Configure’
       i.	Confirm Generator, is set to ‘Visual Studio 15 2017’
       ii.	Specify Platform, select ‘x64’
       iii.	Click Finish and wait for CMake to Configure Build
  d.	Check ‘BUILD CSHARP BINDINGS’
  e.	Click ‘Generate’
5.	Open ‘C:\librealsense-2.19.1\build\librealsense2.sln in Visual Studio 2017
  a.	Right-click ‘Solution librealsense2’ at top of Solution Explorer, select ‘Properties
      i.	For Startup Project, select Current Selection
  b.	In Solution Explorer, select ‘cs-tutorial-2-capture’ (in Wrappers/csharp)
  c.	Generates 203 Warnings, 0 Errors
  d.	Launches application and streams data to Depth and Color image controls 

### To Create, Build and Run WORKING Application from Scratch with v.2.19.1
To create a functional WPF application from scratch, after building Source Files in Visual Studio 
1.	Use Configuration Manager to change Platform from ‘Any CPU’ to <New…> platform ‘x64’
2.	Add a reference to ‘C:\librealsense-2.19.1\build\Debug\Intel.RealSense.dll’
3.	Add using statement ‘using Intel.RealSense;’
4.	Copy ‘realsense2.dll’ from ‘C:\librealsense-2.19.1\build\Debug’ to ‘YourProject\bin\x64\Debug’

### BROKEN C# Wrapper Tutorial in v.2.24.0 and v.2.29.0
Attempting to build and run ‘cs-tutorial-2-capture’ with v.2.24.0 or v.2.29.0 receives error message:
Unable to load DLL ‘realsense2’: The specified module could not be found. (Exception from HRESULT: 0x8007007E)

Observation: 
In v.2.19.1, the DLLs are ‘Intel.RealSense.dll’ and ‘realsense2.dll’
In v.2.24.0 and 2.29.0 the names have changed to ‘Intel.RealSensed.dll’ and ‘realsense2d.dll’. the letter ‘d’ has been added to the end of both dlls in ‘build\Debug’

The same issues apply to creating a working application from scratch

ziyeye, 2019-12-16:
If you run "Release" in "Solution Configuration" instead of "Debug", it creates dll without the letter 'd'.

RealSenseCustomerSupport, 2019-12-19:
----------------------------------------------

Hi @brenteickhoff,

The "d" postfix was added for debug builds. User that doesn't intend to develop/contribute code for the SDK itself should install SDK by following installation guide instructions and build a solution in "Release" mode.
Please let us know if you need any further clarifications regarding this topic.

Thank you!

closed 2019-12-26 by RealSenseCustomerSupport