realsenseai/librealsense #5023: Auto-Exposure stuck when saturated with sunlight on USB 2.0

xerxesb, 2019-10-09:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |  5.11.6.250 | 
| Operating System & Version |   - | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC and OpenEmbedded |
| SDK Version                            |  2.22.0                          | 
| Language                            |  C++                          | 
| Segment			|  Other                          | 

### Issue Description
Like the issues reported in #4140 and #4928, we experience an issue whereby when the IR sensors are exposed to saturating sunlight, the exposure is automatically reduced and becomes stuck there.

According to the comments in #4140, this issue was fixed in firmware 5.11.12.0, however 5.11.12.0 seems to only have been an internal release, not public. The release notes for the latest public release (5.11.15.0) does not indicate that #4140 was resolved.

This is a request to investigate the previously reported issue #4140 and specifically with support for USB 2.0 devices, as we have hardware constraints that prevent us from using USB 3.0

agrunnet, 2019-10-10:
We checked. FW Version 5.11.15.0 should have the fix.  If you experience issues please let us know.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

I just did test between 5.11.6.250 vs 5.11.15.0, and Yes, this issue is fixed in 5.11.15.0 for USB2 mode (actually it was fixed in 5.11.12.0).
Let us know if you see any problem.

Thanks

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.