realsenseai/librealsense #5024: about pipe->start

yoichi0110, 2019-10-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (5.11.15.0) | 
| Operating System & Version |   {Win (10)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.25.0.993 }                          | 
| Language                            |  {C++/C# }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

In SDK, when you run pipe-> start, * operator-> () const _NOEXCEPT is called, but in my project, pipe-> startf will run immediately pipeline_profile start (). What is the difference between this?

lramati, 2019-10-28:
The explicit call to operator->() is likely either being added by your IDE's debugger, or optimized out by the compiler in release mode

lramati, 2019-10-28:
There's no practical difference in terms of program logic that's running

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Thank you @lramati.
Closing this ticket if no further question.

closed 2019-11-15 by RealSenseCustomerSupport