realsenseai/librealsense #5027: Accuracy measure

JonaPR, 2019-10-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { SR305 } | 
| Firmware Version                   | 3.27.3.0 | 
| Operating System & Version |   {Win 10  |                                | 
| Platform                                 | PC  |
| SDK Version                            |  2.28.0                          | 
| Language                            |  {C++ }                          | 

### Issue Description

Hello,

I am measuring the distance between the centers of the circles of the image. I get the pixel center in the color image and use rs2_deproject_pixel_to_point function to get point.

[image]

I have four cameras. The measures between circles' center are different for each camera (the difference is about 4 mm). Is that normal? or there are a way to precalibrate SR305 before use it? I need more accuracy to join four cloud points, but each camera gives to me different measures and I can't get the corretct transformation to apply to my cloud points.

The distance of the cameras to the objects are between 40 and 70 cm

Thanks in advance.

MartyG-RealSense, 2019-10-10:
Where are the cameras placed, please?   Is each of the four cameras placed in the triangles in the center, looking outwards to the circles?

I would expect there to be some difference in the measurements for each camera.  No one camera has exactly the same intrinsics due to differences introduced during the camera's manufacturing process.  

A way to align point clouds is to take the point cloud from every one of the cameras and then do an **Affine Transform**.  Basically, just rotate and move the point clouds, in 3D space, and then once you've done that, you append the point clouds together and just have one large point cloud.

JonaPR, 2019-10-10:
Hi, @MartyG-RealSense , thanks for your answer.

The cameras are placed just behind each circle, looking at the center of the table.

I do affine transform to obtain the large point cloud. But this affine transform it depend of the position of the circles' center. If the posion is not accurate, I can't get a good final point cloud. I need acuracy in the point cloud because I have to get exact measures of the object.

Here is a capture of the position of center from one circle (each color refers to each camera):

[image]

MartyG-RealSense, 2019-10-10:
I would imagine that the accuracy of the measurements on the drawing would be an important factor.   For example with the two small circles, if the cameras were not at the exact same distance from the center point then that in itself could introduce a variation in the distance measurement  Also, if the center square is not at the exact center and measures the same all the way around, the center-point may be further or nearer the camera depending on what side of the center-square the camera is viewing.

JonaPR, 2019-10-10:
The color images I get for the circles are 1920 x 1080. The cameras are placed carefully symmetrical from the center of board. Should I assume that measurement error (around 5 mm) as a cameras tolerance? I would like obtain more accuracy.

MartyG-RealSense, 2019-10-10:
Could you try using a lower color resolution please such as 640x480 and see if it improves your accuracy?

JonaPR, 2019-10-11:
I've been testing 640x480 but it doesn't improve accuracy. Seems to get worse.

MartyG-RealSense, 2019-10-11:
The RealSense Viewer has an **Accuracy** option under the 'Controls' options side-panel section for increasing accuracy by increasing "the number of patterns projected per frame.  The higher the accuracy value, the more patterns projected. Increasing the number of patterns helps to achieve better accuracy. Note that this control affects the depth FPS".

I'm not sure how this function is accessed in code in the current RealSense SDK 2.0 - a member of Intel on this forum could help with advising on that if the RealSense Viewer's accuracy control gives you better results.

JonaPR, 2019-10-11:
I am working with an accuracy value of 3 (maximum). And the maximum frame rate supported too (60FPS with 640x480 or 30FPS with 1920x1080).

MartyG-RealSense, 2019-10-11:
It is possible that you may be at the limits of the SR300 technology.  The SR300 has better accuracy than the 400 Series at very close range, otherwise the 400 Series cameras have higher accuracy.  For example, the 400 Series can achieve less than 1% error under optimum conditions (2% may be more realistic), whilst a figure quoted in the past for the older technology generation the SR300 comes from is around 5%.

JonaPR, 2019-10-11:
I was working before with four cameras d415, but the accuracy wasn't good and I thougth it was becasue I am in a close range (1% or 2% of 500mm is 5 or 10mm, too much for my pretensions) . I decided change the d415 and buy SR305 for that reason. Now I have less noise in the cloud points, but accuracy seems similar to d415.

MartyG-RealSense, 2019-10-11:
SR300 does better than the 400 Series at very close range because it has a smaller scale of depth units (1/32 of a mm).  The 400 Series uses a larger depth scale that helps it to see a greater distance but disadvantages it at very close range compared to the SR300.  You can reduce the 400 Series' depth units scale from its 1 mm default scale to get it closer to the SR300's, but not get it as small as the SR300's scale.

MartyG-RealSense, 2019-10-11:
An approach you could try is to minimize the D415's depth scale as far as it will go, and use the camera's High Accuracy preset configuration too.

JonaPR, 2019-10-11:
I did this before buying SR305. I think I have to assume that I am at the bound of accuracy.

MartyG-RealSense, 2019-10-11:
There is a Short Range preset for the 400 Series camera called ShortRangePreset.json that is not built into the Viewer.  You can download it from the link below.  Visit the highlighted section and right-click on the 'ShortRangePreset' link and select Save Link As (or whatever option your browser has for saving files connected to a link, such as Save Target As).  Once the file is downloaded, you can drag-and-drop it onto the RealSense Viewer window to load it in.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

[image]

As a final resort, you could try turning off auto-exposure and let the Viewer use manual exposure values.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @JonaPR,

SR305 measurement accuracy indeed has its limit even though it's in a closed range. And, the different SR305 has different tolerance per their own factory calibration.
Do you also try different SR305 preset values and see how it works?

JonaPR, 2019-10-31:
Hi @MartyG-RealSense , @RealSenseCustomerSupport ,

I apreciate both of your answers. Yes, I had been testing all these options, but each camera always see the circles in diferent positions (around 5mm). I guess I am at the accuracy bound of the camera, because my range is 40-70 cm, and the 1% of this distance is about 4 and 7 mm.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @JonaPR,

Your result should be within our spec. You probably can have the correlation of each camera's distance measurement with the ground truth and then have its deviation/bias trend for each camera. Consider if the fix of the deviation/bias of each camera can help your use case.

RealSenseCustomerSupport, 2019-12-05:
----------------------------------------------

@JonaPR,
I will close this if no other concern.

closed 2019-12-06 by RealSenseCustomerSupport