realsenseai/librealsense #5029: Intermittent exception when starting `rs-multicam`

evercoast-smm, 2019-10-10:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.18.20-041820-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

We have 5 cameras connected.

We note that when repeatedly starting the `rs-multicam` we see examples, we see the following exception empirically once every 5 runs, and we note that the frequency of the error increases as we add more cameras:
 
```
evercoast@evercoast:~/foo/bar$ rs-multicam 
RealSense error calling rs2_pipeline_start_with_config(pipe:0x55693ef89b00, config:0x55693ef89b20):
    hwmon command 0x10 failed. Error type:  (1).
```

Thank you!

ev-mp, 2019-10-22:
@evercoast-smm , can you provide the kernel log collected when this error occurs ?
Intermittent initialization errors are often an indication of power/bus-related issues.

evercoast-smm, 2019-10-26:
[ 1246.381352] ------------[ cut here ]------------
[ 1246.381355] Unknown pixelformat 0x00000000
[ 1246.381394] WARNING: CPU: 20 PID: 52293 at drivers/media/v4l2-core/v4l2-ioctl.c:1297 v4l_fill_fmtdesc+0xe71/0x1220 [videodev]
[ 1246.381395] Modules linked in: thunderbolt snd_hda_codec_hdmi binfmt_misc nvidia_uvm(OE) intel_rapl nvidia_drm(POE) skx_edac nvidia_modeset(POE) nls_iso8859_1 nfit joydev hp_wmi intel_wmi_thunderbolt sparse_keymap wmi_bmof x86_pkg_temp_thermal intel_powerclamp nvidia(POE) uvcvideo videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 coretemp videobuf2_common videodev media kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc input_leds snd_hda_codec_realtek aesni_intel snd_hda_codec_generic aes_x86_64 crypto_simd cryptd drm_kms_helper glue_helper intel_cstate intel_rapl_perf snd_seq_dummy drm ipmi_devintf snd_seq_oss serio_raw snd_hda_intel snd_hda_codec snd_hda_core ipmi_msghandler snd_seq_midi snd_hwdep snd_seq_midi_event fb_sys_fops syscopyarea snd_pcm snd_rawmidi sysfillrect sysimgblt
[ 1246.381461]  snd_seq snd_seq_device cmdlinepart snd_timer intel_spi_pci intel_spi spi_nor snd mei_me soundcore mtd lpc_ich mei ucsi_acpi typec_ucsi typec ioatdma dca wmi mac_hid sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear hid_generic usbhid hid mpt3sas psmouse raid_class scsi_transport_sas i40e e1000e ahci libahci
[ 1246.381520] CPU: 20 PID: 52293 Comm: rs-multicam Tainted: P        W  OE     4.18.20-041820-generic #201812030624
[ 1246.381523] Hardware name: HP HP Z8 G4 Workstation/81C7, BIOS P60 v01.72 11/29/2018
[ 1246.381534] RIP: 0010:v4l_fill_fmtdesc+0xe71/0x1220 [videodev]
[ 1246.381535] Code: c6 2b d3 16 c1 e9 46 f2 ff ff 48 c7 c6 d4 ce 16 c1 3d 4d 54 32 31 0f 84 34 f2 ff ff 89 c6 48 c7 c7 8b db 16 c1 e8 31 8d d3 d6 <0f> 0b 41 80 7c 24 0c 00 0f 85 36 f2 ff ff 41 8b 44 24 2c 49 8d 7c 
[ 1246.381599] RSP: 0018:ffffb185cf68fc60 EFLAGS: 00010286
[ 1246.381602] RAX: 0000000000000000 RBX: ffffb185cf68fd88 RCX: 0000000000000006
[ 1246.381605] RDX: 0000000000000007 RSI: 0000000000000082 RDI: ffff96d34ef164b0
[ 1246.381607] RBP: ffffb185cf68fc70 R08: 0000000000000001 R09: 0000000000005376
[ 1246.381609] R10: 0000000000000004 R11: 0000000000000000 R12: ffffb185cf68fd88
[ 1246.381611] R13: ffffffffc0c919a0 R14: ffff96cfd91c8300 R15: 0000000000000000
[ 1246.381614] FS:  00007efdc11de780(0000) GS:ffff96d34ef00000(0000) knlGS:0000000000000000
[ 1246.381617] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 1246.381619] CR2: 000056532cf923d8 CR3: 00000014b05ca004 CR4: 00000000007606e0
[ 1246.381621] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 1246.381624] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 1246.381625] PKRU: 55555554
[ 1246.381626] Call Trace:
[ 1246.381640]  v4l_enum_fmt+0x74/0x110 [videodev]
[ 1246.381650]  __video_do_ioctl+0x138/0x2d0 [videodev]
[ 1246.381660]  ? numa_migrate_preferred+0x2b/0x80
[ 1246.381667]  ? task_numa_fault+0x7e0/0xa80
[ 1246.381678]  video_usercopy+0x1a6/0x630 [videodev]
[ 1246.381688]  ? v4l_fill_fmtdesc+0x1220/0x1220 [videodev]
[ 1246.381699]  ? __handle_mm_fault+0x42c/0x5b0
[ 1246.381709]  video_ioctl2+0x15/0x20 [videodev]
[ 1246.381719]  v4l2_ioctl+0x49/0x50 [videodev]
[ 1246.381725]  do_vfs_ioctl+0xa8/0x620
[ 1246.381731]  ? handle_mm_fault+0xe1/0x210
[ 1246.381736]  ksys_ioctl+0x67/0x90
[ 1246.381742]  __x64_sys_ioctl+0x1a/0x20
[ 1246.381747]  do_syscall_64+0x5a/0x110
[ 1246.381752]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[ 1246.381755] RIP: 0033:0x7efdbe2765d7
[ 1246.381756] Code: b3 66 90 48 8b 05 b1 48 2d 00 64 c7 00 26 00 00 00 48 c7 c0 ff ff ff ff c3 66 2e 0f 1f 84 00 00 00 00 00 b8 10 00 00 00 0f 05 <48> 3d 01 f0 ff ff 73 01 c3 48 8b 0d 81 48 2d 00 f7 d8 64 89 01 48 
[ 1246.381820] RSP: 002b:00007ffe1e91a668 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
[ 1246.381824] RAX: ffffffffffffffda RBX: 00007ffe1e91a870 RCX: 00007efdbe2765d7
[ 1246.381826] RDX: 00007ffe1e91a830 RSI: 00000000c0405602 RDI: 0000000000000068
[ 1246.381828] RBP: 000056532e220530 R08: 0000000000000001 R09: 0000000000000008
[ 1246.381830] R10: 000056532cf542e0 R11: 0000000000000246 R12: 00007ffe1e91a740
[ 1246.381832] R13: 0000000000000000 R14: 000000005a313620 R15: 00007ffe1e91aa80
[ 1246.381836] ---[ end trace f34d1118a00e8f7c ]---

RealSenseCustomerSupport, 2019-11-08:
----------------------------------------------

Hi @evercoast-smm,

Please provide us with the following details about your system configuration :
how the cameras are connected to the host machine (direct connection, powered/unpowered hub, etc.)?
are all the cameras running the same FW Version (according to your information it's 05.11.15.00)?
are all the cameras connected to USB2/USB3/mixed/other?

Along with that information, please describe the exact scenario that leads to the problem and specify, inter alia:
what's the minimal amount of cameras connected simultaneously to system causes to above-mentioned issue?
what the streams and their parameters(format, resolution, fps) that cause the issue when they are running simultaneously?

Please note that quality and length of cables are the factors that can cause to the issue. For the best results, we recommend to use short high quality cables.

Thank you!

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

@evercoast-smm,

Do you still need assistance with this matter?

evercoast-smm, 2019-11-15:
Our proprietary utilization of the libRS API results in exactly the same intermittent problems as are exhibited by the 'rs-multicam' example app that's provided with the distribution. Rather than discussing our implementation, we defer to yours and cite problems with it.

All cameras are running the same firmware, 05.11.15.00. All cameras are connected via USB3.2 with adequate bandwidth and power (we have our own internal tools to validate this). We also run height quality USB3 cables and have a mechanism for validating the cable's bandwidth ceiling. The problem arises with, I'd say 4 or more cameras, and as more cameras are added the frequency of the problem increases.

In terms of the streams, format, resolution, fps, etc. We have not modified the 'rs-multicam' example, so it's as configured in your example application.

Our ideal is to have the 'rs-multicam' example application robustly start and terminate. From there we'll follow your best-practice lead.

Thanks very much.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Hi @evercoast-smm,

Please run dmesg -wT in one terminal and rs-multicam in another until you are getting the above mentioned exception and provide us with the contents of the dmesg output.

Thank you!

RealSenseCustomerSupport, 2019-12-20:
----------------------------------------------

@evercoast-smm,

Do you need any further help on this issue?

closed 2019-12-27 by RealSenseCustomerSupport