realsenseai/librealsense #5030: Align to Depth produces lossy color image

delucr, 2019-10-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | 05.10.13.00 | 
| Operating System & Version |   {Win (10)  | 
| Platform                                 | PC |
| SDK Version                            |  {  2.0.<?> }   | 
| Language                            |  {matlab }                          | 
| Segment			|  {others }                          | 

### Issue Description
In following the examples available for using the align function to align the rgb and depth images, all of the examples demonstate aligning depth to color as in:

align_to = realsense.stream.color;
alignedFs = realsense.align(align_to);
.....
 fs = pipe.wait_for_frames();
 aligned_frames = alignedFs.process(fs);

However, if you align color to depth as in:

align_to = realsense.stream.depth;
alignedFs = realsense.align(align_to);
.....
 fs = pipe.wait_for_frames();
 aligned_frames = alignedFs.process(fs);

The result is a color image that contains holes in it at what appears to be the location of zero depth information.  To me this is unexpected.  My understanding of what the align function should do is scale, size, and shift the color image as if it was in the depth camera's coordinate system.  It appears though that the depth image is being used to filter the color image and therefore drop data from the color image where the depth image has zeros.  Using a hole filling filter on the depth before aligning improves the color image but it is sub-standard of what I think align should do.  

Please let me know if this is intended functionality or something that can be fixed.

Best regards,

Rich

ev-mp, 2019-10-15:
@delucr hello,
The term "alignment" w.r.t SDK denotes a synthetic image generation with the help of depth map. In the process a new image is created using triangulation method comprising of 2D=>3D=>2D transformations.
The transformation is akin to a `ray tracing` - each pixel is first "fired" from 2D to 3D space, and then projected into the 2D plane of another sensor.
So by definition the _aligned data is generated from depth in conjunction with color/ir_ data, hence the areas with no depth data coverage result in "holes" in the generated image.

The above serves a specific purpose  - **allowing 1:1 mapping of depth to color and vice versa**. So when a user selects RGB pixel of interest then the SDK will be able to provide the corresponding Depth deterministically with no false positives.
This is a critical feature in scenarios where the depth is essentially used as a validation filter for other sensors -  collision avoidance, segmentation, object recognition.

While using sheer 2D image manipulations would similarly allow to recalculate an image from a different point of view, it would also introduce fundamental flaws - preserve the pixels for which there is no corresponding depth data, and also introduce artifacts due to occlusion and different perspective, which would undermine the declared purpose.

Eventually there are more than one way to "align" images, depending  of the use-case requirements. And the above explanation highlights the specific advantage (and also the utility) of the method implemented by the SDK.

ev-mp, 2019-10-22:
@delucr, do you have any follow ups?
We'll close it if no further questions.

delucr, 2019-10-22:
Thank you for the reply.  I understand why it was implemented the way it was now.  Thanks.

EXTERNAL EMAIL

CAUTION: External email. Do not click links or open attachments unless you recognize the sender and know the content is safe. Please forward all suspicious emails to informationsecurity@stryker.com<mailto:informationsecurity@stryker.com>.

Follow this link to read our Privacy Statement<https://www.stryker.com/content/stryker/gb/en/legal/global-policy-statement.html/>

ev-mp, 2019-10-23:
It was a non-trivial question, you're welcome.

closed 2019-10-23 by ev-mp