realsenseai/librealsense #5031: Is there a way to recognize a person and determine where he or she has a joint?

devLupin, 2019-10-11:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435| 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  Latest                          | 
| Language                            |  C++/C#                          | 
| Segment			|  others                          | 

### Issue Description
<question>

I want to get the angle of a specific joint through image processing. I tried with Kinect sensor but this time I want to do it with RealSense.

My first thought process is as follows:
1. Recognize people through image processing
2. Judge the points of each joint and calculate the three points of the joint to get the angle.

* I don't know how to determine if I am a person through image processing and find out where his joints are. *
Can you please tell me something?

Is there a recommendation method or API? Is there anything I can refer to?

Thank you for your answer.

MartyG-RealSense, 2019-10-11:
The best software available for joint tracking is a commercial package called Nuitrack SDK.  It has a licence fee of $40 a year but has a free trial version.

https://nuitrack.com/

The link below suggests a method for calculating joint angle with Nuitrack.

https://community.nuitrack.com/t/calculate-the-angle-between-two-joints/1121

devLupin, 2019-10-11:
@MartyG-RealSense 

I searched and found it was compatible with the D435. 
Thank you very much. You are my job saver(?).

closed 2019-10-29 by RealSenseCustomerSupport