realsenseai/librealsense #5035: Actual exposure metadata for color frames doesn't update values with autoexposure on

ruminus, 2019-10-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.11.100 | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.15.0-64                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.28.1                          | 
| Language                            |  Python 3.6                         | 

### Issue Description
`actual_exposure` metadata doesn't provide relevant values for **color** frames when autoexposure is enabled. For example, it always returns `166` (default manual value) until you switch to manual, then it will be the new manual value. **Depth** frames return relevant exposure as it auto-adjusts.

MartyG-RealSense, 2019-10-11:
I hope the link below will be helpful to you

https://github.com/IntelRealSense/librealsense/issues/2549#issuecomment-431863920

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Closing this ticket unless there is any further question. Thank you.

closed 2019-11-01 by RealSenseCustomerSupport