realsenseai/librealsense #5038: Compatible USB cables

nacorti, 2019-10-12:
Hey folks,

I've been working through issues with my Jetson Nano using the issued USB 3.0 cable to get connectivity with my D435. I used to easily get 30 FPS off the RGB camera with an 848x480 resolution using the ROS wrapper, but lately I've been noticing degradation to 6-10 FPS at the same resolution, even in Realsense Viewer. And to pre-empt questions, yes I've tried other ports, yes, I'm using the 5W power supply on the Nano, yes, I've done hardware resets and rebuilt the SDK a couple of times.

I believe my issue resembles #4166, #2612, #2045. I'm experiencing a lot of the same symptoms (camera being recognized as USB 2.0, spontaneous disconnects, etc) so I'm 98% sure that I've just bent and twisted the cable around a few too many times and it's messing with my quality

With that in mind 

Can anyone who's got a working 3rd party cable please comment here? Been looking at [NTC](https://www.ntcdistributing.com/usb-3-1-type-c/a-to-c/usb-3-1-a-male-to-c-male-with-dual-screw-locking-cable/?sku=U3S1A01C12-010), [NewNex](https://www.newnex.com/usb-type-c-cables-legacy.php), and [this](https://www.amazon.com/Cablecc-Type-C-Locking-Standard-USB3-0/dp/B076SGCZBS) guy from Amazon. Before I drop $15-50 on cables that may or may not work, would like to see what the community's tried that worked/failed.
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.11.6.250 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9                                         |
| Platform                                 | NVIDIA Jetson Nano  |
| SDK Version                            |  2.25.0                     | 
| Language                            |  ROS/Realsense Viewer                          | 
| Segment			|  Camera Connectivity                          |

MartyG-RealSense, 2019-10-12:
The Newnex and OCP links that you highlighted are good choices, based on the feedback of other RealSense users.  My personal recommendation is Newnex simply because they provide specific RealSense support and provide direct evidence that their cables work with RealSense.

https://www.newnex.com/realsense-3d-camera-connectivity.php

https://youtube.com/watch?v=GLQgR1jT04M&t=13s

Before replacing the cable though, I recommend trying turning off auto exposure to make sure your FPS is not being dragged down by an exposure problem.  

Also look at the ASIC temperature in the RealSense Viewer to check the camera is not overheating - a temperature above 30 from a cold start would be abnormal.  Rapid overheating of a USB device can be caused by a bad USB port or cable, or by a single-board computer like your Nano overheating.

nacorti, 2019-10-13:
Probably going to grab a Newnex based on your recommendation.

Documenting some other issues that I believe stem from the same USB cable issue: From a cold start, Realsense Viewer is registering an ASIC temp of 31 that climbs to 35 after only a couple of minutes turned on. This is with the stereo module only running, set at the recommended 848x480x30 settings. Also, when I try to turn off auto-exposure, I get a pop-up error

`Invalid Value in rs2_get_option(options:0x55abeede40, option:Enable Auto Exp... get_xu(id=11) failed! Last Error: Resource temporarily unavailable`

This is mirrors other issues I have trying to alter settings on the D435. Even changing the preset from "Custom" to "High-Accuracy" causes the whole Viewer application to become unresponsive. Seems like the application doesn't work smoothly if there's not smooth communication with the attached camera, plus I didn't have any of these issues when I first got my cameras.

I should add that I've had no such issues with my T265, which works flawlessly and with minimal latency with both the RealSense Viewer and ROS. 

I'll let y'all know how the new cable works out -- as always, I appreciate the good response and low turnaround time :)

nacorti, 2019-10-13:
One other question: Do you guys have an opinion about the use of externally-powered USB3.1 hubs? Since I'm not sure whether this is a Baud Rate issue or a power issue, I'm not sure whether the extra latency that a hub would add is balanced by the extra power it would supply to the system. This is assuming said hub is used with my Jetson Nano.

MartyG-RealSense, 2019-10-13:
In general, using a powered USB 3.1 hub is a good idea in situations where USB power instability may be occurring.  They are especially useful with single board computers, which have a higher chance of this happening.

nacorti, 2019-10-13:
I'll try out a hub when able. Will upgrade firmware (assuming this won't cause an issue even though I'm running SDK 2.25.0).

As always, thanks for your help! Closing issue

closed 2019-10-13 by nacorti