realsenseai/librealsense #5039: can't get depth image

tonyjonyjack, 2019-10-12:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win (10) | 
| Platform                                 | PC  |
| SDK Version                            |   legacy / 2.0                           | 
| Language                            |  C++/opencv                         | 

### Issue Description
<Describe your issue / question / feature request / etc..>

`
int main()
{
	rs2::colorizer c;
	rs2::pipeline pipe;
	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
	cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
	rs2::pipeline_profile profile = pipe.start(cfg);
	auto depth_stream = profile.get_stream(RS2_STREAM_DEPTH).as<rs2::video_stream_profile>();
	auto color_stream = profile.get_stream(RS2_STREAM_COLOR).as<rs2::video_stream_profile>();
	while (true)
	{
		rs2::frameset frames = pipe.wait_for_frames();
		rs2::frame color_frame = frames.get_color_frame();
		rs2::frame depth_frame = frames.get_depth_frame();
		Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
		Mat depth(Size(640, 480), CV_8UC3, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
		//imshow("1", color);
		imshow("2", depth);
		waitKey(10);
	
	}
	
	return 0;
}`
when I ran codes,I got terrible image ,and seconds later ,program stopped.
[image]

ev-mp, 2019-10-15:
@tonyjonyjack , the corrupted image is due to incorrect `imshow`-ing depth raw data is 16 bit-
>Mat depth(Size(640, 480), **~~CV_8UC3~~**, (void*)depth_frame.get_data(), Mat::AUTO_STEP);

The two immediate methods to fix it:
1) Convert raw depth into colored map, then render it as RGB (#4362). Note that while it is convenient for human presentation, in most cases you will not be able to extract depth from the resulted image.
2. Treat the depth image in OpenCV as native 16 bit data (#3371)
>Mat depth(Size(frame_w, frame_h), **CV_16UC1,** (void *) depth_frame.get_data(), Mat::AUTO_STEP).

Note that while this method will preserve the actual depth,which makes it desirable for CV/Matlab manipulations it is also makes it hard for human interpretation #4289

ev-mp, 2019-10-23:
@tonyjonyjack , did you manage to rectify the issue accordingly?

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@tonyjonyjack Did you get chance to try as ev-mp suggested? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-13:
----------------------------------------------

@tonyjonyjack No feedback from your side for weeks. Will close the issue at this point. Please feel free to create new one if you still have issues/questions. Thanks!

closed 2019-11-13 by RealSenseCustomerSupport