realsenseai/librealsense #5040: No realsense devices were found in ROS Wrapper, but detected by realsense-viewer

studennis911988, 2019-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.11.100| 
| Operating System & Version |   Linux (Ubuntu 16.04 | 
| Kernel Version (Linux Only)    | 4.4.38-tegra                                         | 
| Platform                                 | NVIDIA Jetson TX2  |
| SDK Version                            | 2.29.0                          | 

### Issue Description
Hi, since most jetsonhack repositories for Nvidia TX2 is outdated( [buildLibrealsense2TX])(https://github.com/jetsonhacks/buildLibrealsense2TX) ,so I installed realsense SDK from [LibUVC-backend installation](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md)  , and I am **enable to run the realsense-viewer successfully**.
[image]
But, when I run the [ROS Wrapper](https://github.com/IntelRealSense/realsense-ros), I couldn't find the device. just like the picture showing below.
[image]
Is there any help for this situation? Thanks in advanced!

MartyG-RealSense, 2019-10-14:
The pace of development of the ROS wrapper is not aligned with the release schedule of the main librealsense SDK.  So the wrapper supports librealsense 2.25.0.

studennis911988, 2019-10-15:
Thanks! problem solved!

closed 2019-10-15 by studennis911988