realsenseai/librealsense #5042: T265 fisheye undistort by matlab

hojunson, 2019-10-14:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |T265| 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am wondering it is possible to get undistorted fisheye images in Matlab using given parameters because the model in Matlab seems like different model( Scaramuzza's Taylor model).

MartyG-RealSense, 2019-10-14:
T265 uses a distortion model called Kannala-Brandt.  OpenCV is usually used to undistort the T265 fisheye (in the link below and in the example program for getting depth from fisheye).  I do not know of a means to do it with Matlab unfortunately.

https://github.com/IntelRealSense/librealsense/issues/3880#issuecomment-489049908

RealSenseCustomerSupport, 2019-11-19:
----------------------------------------------

Hi @hojunson

Were you able to utilize the information from MartyG-RealSense to help with your question?  Is there anything further needed here?  As MartyG-RealSense outlined our only example is what is provided at this time.

hojunson, 2019-11-23:
http://www.ee.oulu.fi/~jkannala/calibration/calibration.html

I found this and am trying to utilize it.

closed 2019-12-17 by dorodnic