realsenseai/librealsense #5044: Using a single rs2::syncer to synchronize between devices?

evercoast-smm, 2019-10-14:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.18.20-041820-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_lifetime.md

^^ The above doc states:

```You can also use a single rs2::syncer to synchronize between devices, assuming it makes sense.```

If you could please elaborate on this idea and how it might be utilized? Also, is there any documentation or example code related to the 'syncer' classes?

Thank you!

MartyG-RealSense, 2019-10-14:
There is a user-contributed script here for using rs2::synced with multiple cameras:

https://github.com/IntelRealSense/librealsense/issues/4158#issuecomment-505559462

More documentation on syncer is here:

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0#syncer

dorodnic, 2019-10-15:
As long as frame timestamp domain equals global time (or system time), it should be possible to use syncer to bundle frames from different sensors. It is problematic when using raw hardware clocks.

evercoast-smm, 2019-10-16:
Thank you @MartyG-RealSense and @dorodnic.

Is there a recommended `queue size` for the rs2::syncer? The constructor defaults to 1?

MartyG-RealSense, 2019-10-16:
Have you seen the advice Dorodnic provided yesterday in your frame queue best practices post please?

https://github.com/IntelRealSense/librealsense/issues/5041#issuecomment-542241323

evercoast-smm, 2019-10-19:
We fully understand this now, can close the ticket. Thank you to everyone for your help!

closed 2019-10-19 by evercoast-smm