realsenseai/librealsense #5045: issue building pyrealsense2 following the instructions

bilalghalib, 2019-10-14:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md?source=post_page-----882933271f3d----------------------
<camera D435i
Real Sense Viewer shows my SDK#: 2.29.0 - I am able to get Depth camera video into the realsense viewer 
I'm on Raspberian
Following these instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md?source=post_page-----882933271f3d----------------------

uname -a
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

gcc -v
gcc version 6.3.0 20170516 (Raspbian 6.3.0-18+rpi1+deb9u1) 

cmake --version
cmake version 3.7.2

I am getting the following error at this point in the instructions:
install pyrealsense2
$ cd ~/librealsense/build

for python2
$ cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python)

for python3
$ cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)

>>>>>>>> I get an error here: $ make -j1
Scanning dependencies of target realsense-viewer
[ 85%] Building CXX object tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/model-views.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-6/README.Bugs> for instructions.
tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/build.make:62: recipe for target 'tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/model-views.cpp.o' failed
make[2]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/__/__/common/model-views.cpp.o] Error 4
CMakeFiles/Makefile2:2263: recipe for target 'tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all' failed
make[1]: *** [tools/realsense-viewer/CMakeFiles/realsense-viewer.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

any ideas?
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

ev-mp, 2019-10-15:
@bilalghalib hello,
The specific error - 
>c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,

is not really informative. It could be an indication of a lack of RAM/VirtMem resources, which is often the case with Raspbians. Make sure that you've at least 2Gb Virtual Memory (Swap File), as indicated in the installation guide that you mention above.
I would suggest to make a clean build, including removing the Cmake cache. And then monitor resources utilization with `htop` during build.

Also - please specify the exact RPI model/Distro used, as we do not observe such symptoms with RPI4 running Raspbian Buster 10 with GCC 8.3

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@bilalghalib With the suggestion from ev-mp, did you get through the issue now? Looking forward to your update.Thanks!

RealSenseCustomerSupport, 2019-11-13:
----------------------------------------------

@bilalghalib No feedback from your side for weeks. Will close this at this point. If you still have issues/question, please feel free to open new one. Thanks!

closed 2019-11-13 by RealSenseCustomerSupport