realsenseai/librealsense #5046: flashback of rs-convert

yangliu995, 2019-10-14:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win 10   | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#}                          | 
| Segment			|  {others }                          | 

### Issue Description
<I have built the rs-convert.exe successfully, but every time  when I try to open it,  it just flashback and a txt file created automatically as show in the picture. Some other exe files have the same problem, like rs-color, rs-motion, rs-fw-update...etc. How can I solve this problem? >

[image]

ev-mp, 2019-10-15:
Hi @yangliu995, please run the executable from a command prompt to read the error ( Hold down the `Windows` key and Press `R`;  type `cmd` and hit `Enter` key)

yangliu995, 2019-10-22:
Problem solved. Thanks a lot.

closed 2019-10-22 by yangliu995

nevynwong, 2021-03-13:
@yangliu995 How did you solve it?