realsenseai/librealsense #5049: [pyrealsense2] Unknown Error when Running Pyrealsense2

riven314, 2019-10-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.13-050013-generic                                  | 
| Platform                                 | Lenovo Legion Y740  |
| SDK Version                            |  NA                          | 
| Language                            |  Python                     | 
| Segment			|  NA                          | 

### Issue Description
I created a conda environment and installed pyrealsense2 by ```pip install pyrealsense2 ```. 

I didn't install any SDK 2.x (as I knew my kernel is not LTS version and the related sdk installation to non-LTS kernel is a bit heavy for me). When I tested pyrealsense2 package in Python, I encountered the following error:  

```
Python 3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 19:07:31) 
[GCC 7.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>> pipeline = rs.pipeline()
>>> pipeline.start()\
... 
KeyboardInterrupt
>>> pipeline.start()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Failed to open scan_element /sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0005/HID-SENSOR-200076.2.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
```
I didn't find any related situation in the existing raised issues. How could I settle this? (hopefully without installing SDK 2.x)

ev-mp, 2019-10-15:
@riven314 , the permission error might be due to improperly installed `udev-rules`.
Did you run the script specified in [the installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) ?
```./scripts/setup_udev_rules.sh```

Are you able to run `realsense-viewer` and stream IMU data without `sudo`?

ev-mp, 2019-10-16:
@riven314 hi, the script copies a file named `99-realsense-libusb.rules` to the following location `/etc/udev/rules.d/` and re-triggers the appropriate service to be aware of the new addition. Check whether the file can be found at the designated place and if not - run the script.

To undo the script just delete that `rules` file from the file system and then refresh the service once more ( the exact syntax is in the script file) or reboot.

riven314, 2019-10-17:
@ev-mp , ```pipeline.start()``` works now after I ran ```./scripts/setup_udev_rules.sh```. However, when I proceeded to sample run ```$ python opencv_viewer_example.py```, I encountered the following error:
```
QObject::moveToThread: Current thread (0x55c9970952d0) is not the object's thread (0x55c9970e2d40).
Cannot move to target thread (0x55c9970952d0)

Segmentation fault (core dumped)
```

I ran it in a standalone conda environment. In the conda environment, I installed opencv by PIP (i.e. ```pip install opencv-python```)

ev-mp, 2019-10-22:
@riven314 hi, the error is not related to the SDK -
>QObject::moveToThread..

as we don't use Qt. [Goolge](https://stackoverflow.com/questions/52337870/python-opencv-error-current-thread-is-not-the-objects-thread) suggests that the error is a conflict in the OpenCV distribution shipped with Qt.

SDK-wise the OpenCv examples [require version 3.4](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/opencv#opencv-samples-for-intel-realsense-cameras)

riven314, 2019-10-22:
@ev-mp , Thanks for the suggestion. I have solved the issue now!
I configured a new conda environment. I firstly installed pyrealsense2 by ``` $ pip install pyrealsense2 ```. And then I installed opencv by ``` $ conda install -c conda-forge opencv ```. Dependent version of opencv (version 3.4 in this case) was automatically detected and installed. 
I can run through opencv_viewer_example.py now!

closed 2019-10-22 by riven314