realsenseai/librealsense #5051: About Sync in D435i

big-chan, 2019-10-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |  latest                         | 
| Language                            |  {C++/C# }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

I see [youtube](https://youtu.be/DSm7NYG5bZE) and set slave mode on Viewer. 
No signal is given but the camera still works. Is this working correctly?

MartyG-RealSense, 2019-10-15:
Have you set one of the cameras to '1' (Master) and the other camera to '2' (Slave) in the Inter Cam Sync option, please?

An indication of whether hardware sync is working correctly will be if the timestamps of the cameras drift apart over time.   An extract below from Intel's multiple camera white paper document explains why.

[image]

big-chan, 2019-10-15:
I set '2' (slave) just one camera . I thought the camera wouldn't work if I used slave mode and didn't give a signal, so I wanted to see it not working. If it doesn't work , i will give signal with signal generator. 
But you're saying the camera looks like it's working properly without a signal, right?

MartyG-RealSense, 2019-10-15:
I believe if you are using a signal generator, both cameras are set to '2' (Slave).

If a signal generator is not being used, then one of the cameras set in the Viewer should be '1' (Master) so that it can transmit a trigger signal, and the rest of the cameras are set to 2' (Slave).

Cameras that do not have a working trigger signal will still stream normally.  They are just streaming independently and unsynced.

big-chan, 2019-10-22:
I have one camera, and set it '2' (Slave) in Viewer. And I signal with signal generator. How do i know this works?

MartyG-RealSense, 2019-10-22:
If you have one slave camera responding to a signal generator (and so do not have a second camera to compare timestamps with), perhaps you could point the camera at the visual output of the signal generator to confirm that the timestamp responses of the camera align to what is occurring on the signal display.

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Assuming you configured hardware and software correctly, it should be very easy to check whether your camera is set at Slave mode.
Usually under Slave mode, the camera is very picky to the frequency of the external trigger. (By the way, please keep in mind, when we say multi-cam sync, it is NOT something like GenLock mode. Without trigger pulsing, our Slave cam still keeps running upon its own clock system, independently.) If the external trigger frequency doesn't perfectly match the camera native frame rate, the Slave cam will be seen broken image or flickering image. For example, when you set camera at 30FPS, it actually requires the external trigger Frequency to be set between 29.9966Hz to 29.9984Hz (based on my experiment and the calculation. Yes, the window is very small to users.). With that said, if the external trigger frequency falls out of that range, you will see image broken or flickering.
So typically people won't be easy to see stable video while using an external trigger (from a Function Generator). User must fine tweak the trigger very accurately.
That's why we highly recommend to use 1x camera as the master cam, because the variation of frequency is very small among same model cameras (i.e. D435i or D435).

**Please be informed that we will close this ticket in seven (7) days if we have not heard back from you.