realsenseai/librealsense #5054: CopyVertices does not copy all points

policchio, 2019-10-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415  | 
| Firmware Version                   |05.10.13.00| 
| Operating System & Version |Win 10| 
| Kernel Version (Linux Only)    |                                    | 
| Platform                                 | PC|
| SDK Version                            |2.29.0| 
| Language                            | C#| 
| Segment			| | 

When converting point coordinates with **CopyVertices**, the data is only partially copied. 
This happens both with (first case):
```
var vertices = new float [points.Count * 3];
points.CopyVertices (vertices);
```
And both with (second case):
```
var vertices = new byte [points.Count * 3 * sizeof (float)];
points.CopyVertices (vertices);
```

The code used is the one shown in (C# cookbook):
[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/csharp/Documentation/cookbook.md#generics-support](url)

In the first case, only a total of x, y, z equal to a points.Count is copied into the array (while the values of x, y, z in total are: points.Count * 3).
In the second case even less data are copied (I seem to see that it only copies 76800 points, compared to a DepthFrame resolution of 1280 * 720).

The problem I detected seems to be the same discussed but not solved here:
[https://github.com/IntelRealSense/librealsense/issues/4828](url)

Thanks for your support,
Davide

ogoshen, 2019-10-29:
My bad, problem is here:
https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/wrappers/csharp/Intel.RealSense/Frames/Points.cs#L49

When used to copy vertices this function wrongly assumes that there's a type constraint to structs of `sizeof(float)*3`, so when called on floats it copies 1/3 of the data, when called on bytes 1280&times;720/3/sizeof(float)=76800.

Until this is fixed you can use `var vertices = new Intel.RealSense.Math.Vertex[points.Count];` (or any other matching struct) and if you really need this as a primitive array, use can unsafe code for casting.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

Hi @policchio,

Did you have a chance to try the workaround solution provided by @ogoshen? Please let us know if you need any further help with this matter.

Thank you!

closed 2019-11-14 by RealSenseCustomerSupport