realsenseai/librealsense #5056: Maximum map size T265

neilyoung, 2019-10-15:
Am I right with the assumption, that such a .ply file is the 3D map of the T265? Is there anything known, how big such a map can be?

Thanks

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Hi @neilyoung

I know that you are following the PR link below and have seen the fact that we are doubling the size of the map when that update is merged.  Do you think we should keep this thread open?

https://github.com/IntelRealSense/librealsense/pull/5234

At this stage we keep the details of the map hidden if you will.  The specifics aren't given out.  The map has details that our algo is able to comprehend and understand when imported back into the device.

Thanks

neilyoung, 2019-11-15:
Feel free to close this. I had something different in mind, initially.

closed 2019-11-18 by RealSenseCustomerSupport