realsenseai/librealsense #5057: nodejs-pointcloud.js example throws error when run

Chargeuk, 2019-10-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                                                      | 
| Firmware Version                   | 05.11.15.00                                             | 
| Operating System & Version |   Win 10                                                  | 
| Kernel Version (Linux Only)    |                                                                | 
| Platform                                 | PC                                                           |
| SDK Version                            |  2.0.19.271 (checkout ad1213edaf8caeaefd494b27d81192b28b7b6685)                                   | 
| Language                              |  nodejs                                                     | 
| Segment			        |  others                                                     | 

### Issue Description
When running the checked out (ad1213edaf8caeaefd494b27d81192b28b7b6685) nodejs examples, I saw the nodejs-pointcloud.js example throw the following error when run (no other nodejs example fails in this way):
F:\Dev\libRealSense\checkout\wrappers\nodejs\index.js:1104
      throw new Error(msg);
      ^

Error: error native function rs2_set_option: object doesn't support option #35
    at Object.errorCallback (F:\Dev\libRealSense\checkout\wrappers\nodejs\index.js:1104:13)
    at PointCloud.mapTo (F:\Dev\libRealSense\checkout\wrappers\nodejs\index.js:1735:24)
    at Object.<anonymous> (F:\Dev\libRealSense\checkout\wrappers\nodejs\examples\nodejs-pointcloud.js:43:6)
    at Module._compile (internal/modules/cjs/loader.js:805:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:816:10)
    at Module.load (internal/modules/cjs/loader.js:672:32)
    at tryModuleLoad (internal/modules/cjs/loader.js:612:12)
    at Function.Module._load (internal/modules/cjs/loader.js:604:3)
    at Function.Module.runMain (internal/modules/cjs/loader.js:868:12)
    at internal/main/run_main_module.js:21:11
d:\agent\_work\2\s\src\vctools\crt\crtw32\stdcpp\thr\mutex.c(51): mutex destroyed while busy

RealSenseCustomerSupport, 2019-12-17:
----------------------------------------------

Hi @Chargeuk,

Please run the same example using the latest version of Intel RealSense SDK 2.0 and let us know if
you need further assistance with this matter.

Thank you!

closed 2019-12-24 by RealSenseCustomerSupport

yohanb78, 2020-04-10:
Hello,

I'm trying to code the same program I made in c++ but this time with javascript

I encounter the same issue with the the version 2.33.1. Notice that this problem is not happening with the old version inside the package npm node-librealsense (sdk 2.19.x I believe)

The function mapTo "is working" when I put in comment the lines from 2969 to 2977 of the file  /src/addon.cpp
```
/*
    StreamProfileExtrator extrator(profile);
    CallNativeFunc(rs2_set_option, &me->error_,
        reinterpret_cast<rs2_options*>(me->processing_block_),
        RS2_OPTION_TEXTURE_SOURCE,
        static_cast<float>(extrator.unique_id_),
        &me->error_);
    if (extrator.stream_ == RS2_STREAM_DEPTH) return;
*/

```
But, no color mapping is applied whatever it 's the last or very old sdk version, this function may have an importance with that.

Thank you