realsenseai/librealsense #5063: How to get the actual coordinates of a specific target captured by the camera at a certain resolution relative to the camera？

Dontla, 2019-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435&D435i| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python                           | 
| Segment			|  Robot                       | 

### Issue Description
We have obtained the pixel position of the target obtained from the recognition program, and also obtained the depth (Z) of the target through the alignment of the depth map to the color map. Next, how can we obtain the actual target through the pixel position of the target? That is, relative to the X and Y coordinates of the camera?

Best regards,

Dontla

ev-mp, 2019-10-16:
@Dontla , converting a depth 2D pixel into 3D vertex is achieved by calling 
[`rs2_deproject_pixel_to_point`](https://github.com/IntelRealSense/librealsense/blob/master/include/librealsense2/rsutil.h#L67) API.
Note that if you're using an "depth aligned to xxx" image then the intrinsic that you need pass to the function must be taken from the "aligned" and not the original depth frame.

Dontla, 2019-10-17:
@ev-mp Thanks Thank you for providing the requested information. But I wonder if you can provide the code implemented with Python, because sometimes I can't find the code repositories or packages corresponding to Python from the code of C/C++.

ev-mp, 2019-10-22:
@Dontla hi, there are [some examples](https://github.com/IntelRealSense/librealsense/search?l=Python&q=deproject) available for python that utilize the API.
Let us know if this helps.

RealSenseCustomerSupport, 2019-11-04:
----------------------------------------------

Hi @Dontla,

Do you need any additional assistance with this question?

Thank you!

closed 2019-11-08 by RealSenseCustomerSupport