realsenseai/librealsense #5064: E: Unable to locate package intel-realsense-dfu*

vrmn, 2019-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i |
| Firmware Version                   | 5.11.1  | 
| Operating System & Version   | Ubuntu 18.04 | 
| Kernel Version (Linux Only)    | Linux 4.15.0-1048-raspi2 aarch64 | 
| Platform                                  | Raspberry Pi 3 B+  |
| SDK Version                           | librealsense2 SDK | 
| Language                               |  n/a | 
| Segment			         |  n/a | 

### Issue Description
Please help I know this is not a new issue but it is one I seem to not find the solution for even after multiple tickets have been closed. 

The command sudo apt-get install intel-realsense-dfu* still won't work even after taking into consideration the spacing errors in the Linux-Realsense-D400-DFU-Guide.pdf from copy and pasting. 

I've also tried just running sudo apt-get install intel-realsense-dfu without the * and still nothing.

I also tried the available docker image with the preinstalled Linux DFU only to get as an output: standard_init_linux.go:211: exec user process caused "exec format error". Probably because of possible hardware limitations mentioned in within the comments of issue #3193. Is there anything I can do to finish the install of the DFU tool to update the realsense firmware in order to connect the D435i camera.

 Thank you.

MartyG-RealSense, 2019-10-16:
Intel changed to a new cross platform firmware updated system this year.  You can now update with a tool called **rs-fw-update** or from within the RealSense Viewer using the More option at the top of the options side-panel.

https://dev.intelrealsense.com/docs/firmware-update-tool

If you choose to use the Viewer, selecting the More icon's menu option to update the firmware to the recommended version will update the camera automatically without you having to download a firmware file.

vrmn, 2019-10-16:
Ok so there might be an issue with the RPI 3 B+ 2.0 ports. I tried updating the firmware using the Viewer screenshots of what I've encountered down below:

[image]
[image]
[image]

So then I try the rs-fw-update tool and this is what I encountered down below:

[image]

Just in case i switched the serial number

[image]

And even tried rs-fw-update -f Signed_Image_UVC_5_11_15_0.bin since I only had the D435i cam connected. Still the same output.

[image]

Is it a 2.0 port problem and if so is there away around this issue?

Thank you

MartyG-RealSense, 2019-10-16:
I am able to update the firmware of my own D435i in the Viewer on a USB 2.0 port, so USB2 in general should work for the Viewer's firmware update mechanism.  I am on Windows though so have not personally experienced firmware update errors in Linux.  One of the RealSense team members on this forum is better equipped to advise on this particular question.  I do apologise.

dorodnic, 2019-10-17:
Firmware update in USB2 mode can be problematic. 
Since firmware update relies on read-only part of the camera firmware (the bootloader) if the firmware was not capable of updating in USB2 mode when the device was *manufactured* this will not be possible at any later time. Since USB2 update was introduces relatively recently this creates a situation where some devices can be updated via USB2 and some not. 
@vrmn - do you have an option to update on a different PC using USB3?

vrmn, 2019-10-18:
@dorodnic hmmm maybe I'll set up realsense on my laptop and see if updates the firmware there and hopefully that helps for the Pi. Thanks man and thanks @MartyG-RealSense for some valuable tips as well.

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Hi @vrmn,

Did you have a chance to try firmware update using USB 3? Please let us know if you need any
further help with this question.

Thank you!

closed 2019-11-05 by RealSenseCustomerSupport