realsenseai/librealsense #5068: T265: What is the purpose of export/import map?

neilyoung, 2019-10-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  Buster Lite, latest
| Kernel Version (Linux Only)    |      | 
| Platform                                 |Raspberry P  |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |  python                      | 
| Segment			|  others                           | 

### Issue Description

Not an issue. A question. Can somebody please explain, what the purpose of "export/import" map in realsense viewer (and the adequate SDK functions is)? Except that there is some file stored on the disk (which can't be found afterwards due to the missing .raw extension) I see no visible effect of especially importing the map again. My hope was, that I would be located somewhere "on the recently map" when I reload a map and resume tracking.

ev-mp, 2019-10-22:
@neilyoung , the way to interact with previously generated (and then imported) maps is by marking and recording "anchor" points and later querying their position with regards to the current map. 
`set/get_static_node`  [APIs](https://github.com/IntelRealSense/librealsense/wiki/API-Changes#t265-specific-apis) are utilized in the use-case.  

Follow the explanation by @radfordi [here](https://github.com/IntelRealSense/librealsense/issues/4024#issuecomment-496025822)

neilyoung, 2019-10-22:
@ev-mp I see. Yes, I already have seen this post. Excuse me, but my understanding of the use of a map was this so far:

- Create it while tracking - that's what SLAM is for
- Make it available for down/upload
- Later upload it to the cam in order to **immediately** locate me at the point in the room, where I'm standing. So the cam should not start again from 0,0,0, but the first coordinate should be where I am
- Make the map more accurate over the time with further mapping.

Would be great, if a behaviour like this is on Intel's agenda for the future. 

Also - It would be great, if the AprilTag detection could be a part of the core code, so that detected tags would be presented like poses. This is especially important for battery driven devices. Right now I'm able to have 3h of pose delivery on a RPI 3B+ (w/o Wifi/BT optimization so far). I fear that adding the cam stream to this would not improve my energy bilance.

BTW: Is that AprilTag format also able to carry some customized textual information, which you would be able to detect?

ev-mp, 2019-10-22:
@neilyoung , the use-case you describe is a highly anticipated feature that was previously requested in several posts. I'll tag @radfordi and @bfulkers-i  to see if they can contribute to this conversation.

neilyoung, 2019-10-22:
@ev-mp Yepp, already seen that too :) This would make the cam really awesome...

TheBricktop, 2019-10-22:
This is the most needed feature that would really differentiate it from any other api on the market.

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Wanted to notify the thread that there has been some updates with regards to maps and relocalization in the latest release:  https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2310

neilyoung, 2019-12-13:
This can be closed, since the required functionality is part of the SDK

closed 2019-12-13 by neilyoung

radfordi, 2019-12-13:
Thanks for following up, @neilyoung.

neilyoung, 2019-12-13:
@radfordi You’re welcome, Jim. Keep on doing your great job!