realsenseai/librealsense #5070: Rename Realsense "Viewer" to "Editor" as it makes changes on the camera settings

tispratik, 2019-10-16:
### Issue Description
The realsense viewer makes changes on the camera settings which persist for a long time until the camera is unplugged and replugged. This creates problem when someone inadvertently changes something in the "viewer" while just trying to "view" and "interact" with the camera. Then when we run our python program, it starts running on the new settings and things don't work as they worked prior to the settings change.

Even if we had to make sure the python program sets the critical settings, there is still no master list of settings available that the "viewer" changes. This is causing problems for us as the system becomes unpredictable for production deployments.

neilyoung, 2019-10-16:
Interesting. Thanks for reporting.

dorodnic, 2019-10-17:
Hi @tispratik 
Thank you for the feedback.
By design the Viewer does not change any settings without explicit command from the user. 
IMHO if merely opening the Viewer and starting streaming is affecting some setting that persists it can be a firmware / software bug.

tispratik, 2019-10-17:
Let's say we change the exposure setting using the viewer and close the viewer. Then run a python program to stream the camera, the python program will be using the new exposure setting that we set through the viewer.

dorodnic, 2019-10-24:
I think this is consistent to other camera "viewers". If you change one of the controls, new setting persists. Some cameras "reset" on power cycle, some keep these settings persistent in camera memory. 
While I understand your concerns, these don't justify product rename in my opinion.

tispratik, 2019-10-24:
We got tripped on this and discovered the issue a few months later. It would have saved us quite a lot of troubleshooting and hair pulling experience. I would say, lets be more explicit and warn the users at least.

dorodnic, 2019-10-27:
I understand. We'll implement something to address it.

closed 2019-12-12 by RealSenseCustomerSupport