realsenseai/librealsense #5071: PointCloud.Process only returns part of the values

tomloomans88, 2019-10-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.22.0.856 }                          | 
| Language                            |  C# | 
| Segment			|  others            | 

### Issue Description
I'm trying to store a point  cloud in c#, but the PointCloud.Process only fills  1/3rd of the actual values, the rest is zero for some reason...

Here is the code I'm using:

`
using Emgu.CV;
using Emgu.CV.Structure;

var ctx = new Context();
            var cfg = new Config();
            cfg.EnableStream(Intel.RealSense.Stream.Depth, 1280, 720, Format.Any);
            cfg.EnableStream(Intel.RealSense.Stream.Color, 1280, 720, Format.Bgr8);
            var pipeline = new Pipeline(ctx);
            var profile = pipeline.Start(cfg);

            using (var frames = pipeline.WaitForFrames())
            {
                //  Get  pointcloud
                var depth = frames.DepthFrame;
                var colorFrame = frames.ColorFrame;
                var pc = new PointCloud();
                var points = pc.Process(depth).As<Points>();
                var vertices = new float[points.Count * 3];
                points.CopyVertices(vertices);

                // Make x, y, z images
                var valuesX = new float[points.Count];
                var valuesY = new float[points.Count];
                var valuesZ = new float[points.Count];

                var height = depth.Height;
                var width = depth.Width;

                for (int row = 0; row < height; row++)
                {
                    for (int column = 0; column < width; column++)
                    {
                        var index = (row * width + column);
                        valuesX[index] = vertices[index * 3];
                        valuesY[index] = vertices[index * 3 + 1];
                        valuesZ[index] = vertices[index * 3 + 2];
                    }
                }

                //  Store images
                unsafe
                {
                    fixed (float* p = valuesX)
                    {
                        IntPtr ptr = (IntPtr)p;
                        Image<Gray, float> imageX = new Image<Gray, float>(width, height, width*4, ptr);
                        imageX.Save("X.tif");
                        // do you stuff here
                    }
                }

                unsafe
                {
                    fixed (float* p = valuesY)
                    {
                        IntPtr ptr = (IntPtr)p;
                        Image<Gray, float> imageY = new Image<Gray, float>(width, height, width * 4, ptr);
                        imageY.Save("Y.tif");
                        // do you stuff here
                    }
                }

                unsafe
                {
                    fixed (float* p = valuesZ)
                    {
                        IntPtr ptr = (IntPtr)p;
                        Image<Gray, float> imageZ = new Image<Gray, float>(width, height, width * 4, ptr);
                        imageZ.Save("Z.tif");
                        // do you stuff here
                    }
                }

                unsafe
                {
                    Image<Gray, UInt16> imgDepth = new Image<Gray, ushort>(depth.Width, depth.Height);
                    UInt16* data = (UInt16*)depth.Data;
                    for (int row = 0; row < height; row++)
                    {
                        for (int col = 0; col < width; col++)
                        {
                            imgDepth.Data[row, col, 0] = (ushort)(data[row * width + col]);
                        }
                    }
                    imgDepth.Save("depth.tif");
                }

                Image<Bgr, byte> imgColor;
                using (Bitmap image = new Bitmap(colorFrame.Width, colorFrame.Height, colorFrame.Stride, System.Drawing.Imaging.PixelFormat.Format24bppRgb, colorFrame.Data))
                {
                    imgColor = new Image<Bgr, byte>(image).DisposeWith(frames);
                    imgColor.Save("color.tif");
                }
            }`

Also if I look into the values in the vertices, after 1/3rd of the rows they are all zero. The depth image is correct and the color image also. I hope somebody can help me with this problem.

[image]
[image]

BryanEmmerson, 2019-10-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |  n/a                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.29.0                         | 
| Language                            |  C#                          | 
| Segment			|  Robot                          | 

### Issue Description
Same as @tomloomans88: I cannot get more than exactly 1/3 of the data when using the C# wrapper. I currently have to write a PLY file to disk, then open this file to capture a complete depth frame.

Any suggestions on this would be much appreciated.

ev-mp, 2019-10-22:
Similar to #5054

RealSenseCustomerSupport, 2019-12-13:
----------------------------------------------

Should be addressed in #5054.
Closing this ticket.

closed 2019-12-14 by RealSenseCustomerSupport