realsenseai/librealsense #5072: unable to install pyrealsense

aswinkumar2019, 2019-10-16:
I tried to install pyrealsense in my jetson nano using pip install pyrealsense
The installation stop with an error no such file or directory rs.h
#include<rs.h>
I have librealsense installed in my device.
Please help me to fix this

g2-bernotas, 2019-10-17:
Just a guess, but maybe?
```
pip install pyrealsense2
```

aswinkumar2019, 2019-10-17:
I tried this and got message as, no such package found.
That's why I tried this. 
It works perfectly in my Ubuntu PC but, doesn't work in my jetson nano

JvanZijl, 2019-10-17:
Pip install of pyrealsense is not possible for ARM architectures. You should be able to build it from source though.

aswinkumar2019, 2019-10-17:
I am not able to build from source too.
It does not work on both my jetson nano 
I tried to use pip install pyrealsense on raspberry pi3 too, first I tried pip install cython. 
It stops after collecting the package and,there is no response. 
Has anybody successfully installed pyrealsense on jetson nano or raspberry pi 3????
If not, is there any docker image for armhf version which has pyrealsense????

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

@aswinkumar2019 Did you try build with source code according to https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source ?

And what's your Ubuntu version on Jetson Nano? Here I tried the build from source code with Ubuntu18.04 on Jetson Nano and didn't get issue.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@aswinkumar2019 Any update from your side? Looking forward to your reply. Thanks!

aswinkumar2019, 2019-11-15:
Sorry for late reply,
Yeah i have ubuntu 18.04
I got these errors during cmake 

CMake Error at third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

  Traceback (most recent call last):

    File "<string>", line 1, in <module>
    File "/usr/lib/python3.6/distutils/sysconfig.py", line 14, in <module>
      import re
    File "/usr/lib/python3.6/re.py", line 142, in <module>
      class RegexFlag(enum.IntFlag):

  AttributeError: module 'enum' has no attribute 'IntFlag'

  Error in sys.excepthook:

  Traceback (most recent call last):

    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 53, in apport_excepthook
      if not enabled():
    File "/usr/lib/python3/dist-packages/apport_python_hook.py", line 24, in enabled
      import re
    File "/usr/lib/python3.6/re.py", line 142, in <module>
      class RegexFlag(enum.IntFlag):

  AttributeError: module 'enum' has no attribute 'IntFlag'

  

  Original exception was:

  Traceback (most recent call last):

    File "<string>", line 1, in <module>
    File "/usr/lib/python3.6/distutils/sysconfig.py", line 14, in <module>
      import re
    File "/usr/lib/python3.6/re.py", line 142, in <module>
      class RegexFlag(enum.IntFlag):

  AttributeError: module 'enum' has no attribute 'IntFlag'

Call Stack (most recent call first):
  third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@aswinkumar2019  You current issue seems to be python 3.6 related issue.  Please check  https://stackoverflow.com/questions/43124775/why-python-3-6-1-throws-attributeerror-module-enum-has-no-attribute-intflag to see if helps.

aswinkumar2019, 2019-11-15:
I had already tried those solutions.Each solution gives rise to new issue.
The problem is, i didn't run it from top of directory.
I ran it inside python wrappers directory which caused the problem.
Please mention it clearly in installation instructions.

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@aswinkumar2019 Sorry for the trouble caused by unclear. And does it mean that the issue solved now? Thanks for the feedback.

aswinkumar2019, 2019-11-22:
Yeah the issue is solved

closed 2019-11-22 by aswinkumar2019

beyond21299, 2019-12-30:
@aswinkumar2019 Can you tell me how to solve the problem during the installation of pyrealsense2？
I have installed librealsense SDK2 on TX2, but I cannot install pyrealsense.
I tried to download the pyrealsense source code and tried 'python3 setup.py install', which caused your initial problem of " not found 'rs.h'".

russ@russ-desktop:~$ sudo find / -name rs.h 
[sudo] password for russ: 
/usr/local/include/librealsense2/rs.h
/home/russ/Downloads/librealsense-master/include/librealsense2/rs.h
/home/russ/.local/share/Trash/files/rs.h
/home/russ/librealsense_build/librealsense-master/include/librealsense2/rs.h

Can you tell me how you solved it?

aswinkumar2019, 2019-12-31:
You should not manually install using setup.py.
You just have to run cmake at the root of the directory.
Hope this helps.
Regards,
Aswin Kumar

aswinkumar2019, 2019-12-31:
BTW you should not install using using pip.
you should compile from source by running cmake at the root of the directory