realsenseai/librealsense #5074: Raspberry Pi 4 Can't Work With D435i through usb 3.0,But Can Run With T265

xilunaJumma, 2019-10-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Rasbian  | 
| Kernel Version (Linux Only)    |  (4.19)                                         | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hey!
I can use t265 with realsense-viewer through USB3.0 and USB2.0,But It can not work with D435i through USB3.0,just work through usb2.0.
When I run realsense-viewer,it shows 
WARNING [3069628976] (ds5-factory.cpp:891) DS5 group_devices is empty.

dorodnic, 2019-10-17:
Hi @xilunaJumma 
Are you building with `-DFORCE_LIBUVC=true` CMake flag? Are you building SDK version >= 2.26? 
These should be the requirements for getting D435i working on the Pi, plus @ev-mp submitted pull-requests recently with small fixes for the Pi that will be included in the next formal release.

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

Please let us know if you have further question question on this topic. Thank you.

closed 2019-11-15 by RealSenseCustomerSupport

Huibean, 2020-09-26:
is this really solved?