realsenseai/librealsense #5075: Convert .bag file in .ply file Realsense Android

nafidoni, 2019-10-17:
Hello Guys,
I want to convert a .bag file to a point cloud .ply type in realsense Android.
What should I do?
Does anyone can help?

Thanks.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

@nafidoni Sorry, currently we don't have such JAVA API yet. If native API also acceptable for you, you can refer to C++ rs-convert to write one native API. https://github.com/IntelRealSense/librealsense/tree/master/tools/convert

RealSenseCustomerSupport, 2019-11-15:
----------------------------------------------

@nafidoni  Does native API work for you? Any other questions? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@nafidoni  Could you please update? Thanks!

nafidoni, 2019-11-22:
@RealSenseCustomerSupport No it didn't work.

Thanks for your help.

RealSenseCustomerSupport, 2019-11-25:
----------------------------------------------

@nafidoni  Could you please elaborate more why you need this function? Can non-real-time mode be acceptable for you? Thanks!

RealSenseCustomerSupport, 2019-12-23:
----------------------------------------------

@nafidoni Could you please update so that we can further look at this? Thanks!

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

@xilunaJumma Sorry that we didn't get response from you for weeks. Will close this issue at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-02-03 by RealSenseCustomerSupport