realsenseai/librealsense #5078: T265: realsense-viewer (compiled from latest SDK) scale issue and crash

neilyoung, 2019-10-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.1.0.279
| Operating System & Version |  Buster Lite, latest
| Kernel Version (Linux Only)    |      | 
| Platform                                 | Ubuntu 16.04  |
| SDK Version                            |   2.2.9.0                           | 
| Language                            |                     | 
| Segment			|  others                           | 

I have recorded a 650 m loop drive "around the block" from the dashboard of my car. While trying to display it and analyze the trajectory with realsense-viewer I'm appearing two issues:

1) The trajectory display is pretty wide and I need to zoom out. After a certain zoom step not only the grid disappears, but also the entire path. I'm attaching a screen recording later.

2) The entire app get's stuck on attempt to pause/skip/resume. The console log says:

```
decades@ubuntu:~/Documents/Python/intel$ ./realsense-viewer 
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
 17/10 15:16:03,089 ERROR [140177339512640] (playback_device.cpp:305) Error - timeout waiting for pause, possible deadlock detected
realsense-viewer: /home/decades/Documents/librealsense/src/media/playback/playback_device.cpp:306: void librealsense::playback_device::pause(): Assertion `0' failed.
Aborted (core dumped)

```

neilyoung, 2019-10-17:
Here is a GIF. I'm zooming out. After the first disappearance I'm zooming in again and so on.

[image]

neilyoung, 2019-10-22:
Is there anything on my side, which could be done in order to allow the scaling of large extents?

dorodnic, 2019-10-22:
Hi @neilyoung 
I think this is a UI issue in the Viewer, caused by 3D projection matrix (near & far clipping planes setup). 
We should look into it, but at the moment the options are:
a. Using dedicated visualisation script for this use-case
b. Using one of the ROS tools, such as rviz. These are also open-source and are slightly more mature, compared to RealSense specific tools. The downside is rviz not being cross platform **yet** as far as I know.

neilyoung, 2019-10-22:
@dorodnic Thanks. Not such a big issue. I think I will give the ROS tools a shot. Thanks for the hint.

neilyoung, 2019-10-24:
@dorodnic I'm now having a full installation of the ROS tools on my Linux laptop, but I can't figure out, how to display the bag. I'm able to replay it (rosplay), but there is a knowledge gap, how to visualize it. Do you know the command line?

neilyoung, 2019-10-31:
Anybody having a hint? I can playback, but I can't visualise anything. Samples also don't refer on playback of ROS recorded files only.

neilyoung, 2019-11-01:
@dorodnic Hey, common. You can't give such suggestions like 

> Using one of the ROS tools, such as rviz. 

and then disappear w/o telling, _HOW_ do view an "foreign" ROS bag. I would have assumed, you have done that before you give such suggestions. Did you? I doubt.

neilyoung, 2019-11-08:
@dorodnic Would it be possible to give me a little pointer, how to view a T265 rosbag with the ROS viewer tools? Otherwise this issue can be closed.

RealSenseCustomerSupport, 2020-01-27:
----------------------------------------------

@neilyoung

Can this be closed out?

neilyoung, 2020-01-27:
No. Still alive.

RealSenseSupport, 2020-09-15:
Thank you for highlighting this issue. We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

neilyoung, 2020-09-16:
@RealSenseSupport Yeah, thank you. Did get something like this from my local car dealer too "...we have moved our focus to our next generation of products. Use a bicycle if you can't afford our shining e-Cars anymore"....

closed 2020-09-23 by RealSenseSupport