realsenseai/librealsense #5079: T265 not recognized on Raspberry Pi 4 if connected at boot time

lionkov, 2019-10-17:
When the Raspberry Pi 4 boots, rs-enumerate-devices doesn't list the connected T265 camera. If the camera is unplugged and plugged again, it is recognized. 

Running with modified kernel 4.19.49 that includes the patches from the scripts directory.

Any ideas how to fix this?

MartyG-RealSense, 2019-10-17:
Problems with T265 being correctly detected on boot is a known issue that the RealSense team are aware of.

https://github.com/IntelRealSense/librealsense/issues/4681#issuecomment-538537601

A different issue regarding Pi 4 and the IMU is detailed here:

https://github.com/IntelRealSense/librealsense/issues/4979#issuecomment-540456439

RealSenseSupport, 2020-09-15:
Thank you for highlighting the recognition issue on Raspberry Pi with regards to T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-17 by RealSenseSupport