realsenseai/librealsense #5080: D435i can not use Dynamic Calibrator app

guyuehb, 2019-10-18:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense USB2|
|**Serial Number**|841612071398|
|**Firmware Version**|05.10.13.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|850123050850|
|**Firmware Update Id**|850123050850|

Please provide a description of the problem

MartyG-RealSense, 2019-10-18:
I note that you are using USB2.  The dynamic calibrator tool was written for USB 3, as USB 2 support was not introduced to the 400 Series cameras until later after its launch.

RealSenseCustomerSupport, 2019-10-31:
----------------------------------------------

Hi @guyuehb,

Did you have a chance to try dynamic calibrator tool using USB 3? Please let us know if you need any
further help with this question.

Thank you!

guyuehb, 2019-10-31:
After that,I had tried using USB 3.Then it has a new error,
"Error: calibration tables on device are not supported, Please try again with latest Dynamic Calibrator software. If you are already using latest software, please re-calibrate the device with latest OEM Calibration tools and try again."
My dynamic calibration version is 2.6.8.0 and firmware version is 05.11.15.00.

MartyG-RealSense, 2019-10-31:
If your camera has a corruption of the 'calibration table' stored inside it, this can be repaired with the CustomRW tool of the Dynamic Calibrator tool by doing a "gold reset" of the camera to factory defaults.  Instructions for doing so are in the link below.

https://forums.intel.com/s/question/0D70P000006EFHESA4

RealSenseCustomerSupport, 2019-11-01:
----------------------------------------------

@guyuehb,

Please let us know if resetting calibration to default gold factory settings has solved the issue and whether or not you need additional assistance with this matter.

Thank you!

closed 2019-11-08 by RealSenseCustomerSupport