realsenseai/librealsense #5081: In python, are the depth frames playbacked from the.bag file sequentially?

nevynwong, 2019-10-18:
| Camera Model                       |  D415 | 
| Operating System & Version |   Linux (Ubuntu 16)  |  
| Language                            | python  | 

I want to get the sequence of frames of the bag file through playback, but does wait_for_frames get the sequence of frames?

g2-bernotas, 2019-10-18:
I believe wait_for_frames should give you one frame at a time. How about you add frames one by one to a list of some sort - just repeat a loop for as many frames as you require in your sequence. For example:

``` Python
framesSeq = []
for _ in range(5):
    framesSeq.append(pipe.wait_for_frames())
```

nevynwong, 2019-10-18:
Hi @g2-bernotas ,
I use a  loop to extract all the required frames, for example:

for I in range (100) :

     frame = pipe. Wait_for_frames ();

     * * * * * *

And then every frame is processed, but it feels like every frame is the same.

ev-mp, 2019-10-29:
Similar to #5144, #5107, #5096

closed 2019-10-30 by nevynwong