realsenseai/librealsense #5082: pipeline start

yoichi0110, 2019-10-18:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

When pipe-> start is executed from the main function, it ends normally, but after executing Application.Run, an exception occurs at pipe-> start. Why?

ieperen3039, 2019-10-22:
what do you do when you say "executing Application.Run"? What language are you using? Do you use an IDE? 
Also, what error do you get? (don't paste the entire log, just the error message)

yoichi0110, 2019-10-23:
application.run is a routine that performs processing to execute an application in visual studio .net. The IDE uses Visual Studio. Then, the exception of the error is that the camera is already streaming when pipe-> start is executed.

yoichi0110, 2019-10-23:
The language is C #.

RealSenseCustomerSupport, 2019-10-30:
----------------------------------------------

Perhaps the C# examples from RealSense SDK could help?
https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp
Please let us know if you have any further question. Thank you.

closed 2019-11-07 by RealSenseCustomerSupport