realsenseai/librealsense #5085: Wrangling D435i pointclouds and T265 pose coordinate systems

MoBaT, 2019-10-18:
Hey!

I'm having trouble wrangling the two coordinate systems into NED space and plotting the Depth Point Clouds and Trackers pose in World Coordinates in NED space. Is what I'm doing below correct? :

1. Convert T265 coordinate system into NED by multiplying the tracker's position and orientation by Quaternionf(0.5, 0.5, 0.5, -0.5) (90 degrees along X, -90 degrees along Z)
2. Convert D435i DepthPtCloud into NED by multiplying the points by Quaternionf(0.5, -0.5, -0.5, -0.5) (-90degrees along X, -90 degrees along Z)
3. DepthPtCloudInNEDGlobal = (T265rotation * D435iPtCloud) + T265position

ev-mp, 2019-10-22:
@MoBat hello,
T265 is "inside-out" tracking system and it cannot know its (full) position in world CS. 
For tracking you'll need to associate T265 initial position with the world's [x,y,z,h] coordinates (T265 is aware of the "world"'s pitch and roll angles).
Obtaining [x,y,z] is relatively straight-forward with a help of GPS or a similar system.
Magnetic sensor can be used to establish initial heading w.r.t real north .

Related to #5009

MoBaT, 2019-10-25:
@ev-mp I'm a bit confused. Can't the tracker's position and orientation be considered the "world" and I would need to shift the point cloud data coming out in body coordinates into the tracker's world?

ev-mp, 2019-10-27:
NED is a specific term used for geo-referencing, where you track one's position w.r.t an external frame (World position).
In case you don't have this requirement, and need to track the device internally, like in VR setting, then indeed you need to multiply the pointcloud by T265 orientation matrix.

Try this in `realsense-viewer` - activate both T265 and D435, switch to 3D view and try walk  - you'd see that the 3D pointcloud's origin from D435 is tethered to the position reported by T265.
Internally, it is achieved by first moving the openGl rendering matrix into the position defined by T265, and then rendering the pointcloud. The sample code is available [here](https://github.com/IntelRealSense/librealsense/blob/d19829788008b8e000870895a068f0c43d58895a/examples/example.hpp#L825)

RealSenseCustomerSupport, 2019-11-14:
----------------------------------------------

Hi @MoBaT - ev-mp provided some good information regarding tethering D435 pointcloud to T265 position as well as provided link to the sample.  I think this can be closed now.

closed 2019-11-20 by RealSenseCustomerSupport