realsenseai/librealsense #5088: Align depth image to color image, but not align in the frame set

SirNightTW, 2019-10-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 5.11.10.0 | 
| Operating System & Version |   Win 10 | 
| Kernel Version (Linux Only)    |  | 
| Platform                                 | PC |
| SDK Version                            |   | 
| Language                            | python | 
| Segment			|   | 

### Issue Description
I'm using D435 and python wrapper.

I want to align depth to color image in 60fps, but when I use the align function(5th and 8th line), the fps drops to about 20, if I don't align them, the fps can reach 60.

rs means pyrealsense2, np means numpy

**1. pipeline = rs.pipeline()
2. config = rs.config()
3. config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60)
4. config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60)
5. align = rs.align(rs.stream.color)
6. profile = pipeline.start(config)
7. frames = pipeline.wait_for_frames()
8. frames_aligned = align.process(frames)
9. color_frame = frames_aligned.get_color_frame()
10. depth_frame = frames_aligned.get_depth_frame()
11. color_image = np.asanyarray(color_frame.get_data())
12. depth_image = np.asanyarray(depth_frame.get_data())**

To solve this problem, I want to save color and depth image to .png and .raw separately, and align them afterwards, rather than align depth and color when they are in a frame set.

**frames = pipeline.wait_for_frames()
color_frame = frames.get_color_frame()
depth_frame = frames.get_depth_frame()
color_image = np.asanyarray(color_frame.get_data())
depth_image = np.asanyarray(depth_frame.get_data())
→ Then align depth_image to color_image**

Is there any solution to my problem?

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

There is high CPU load when align is working.
Please try align the frames offline (calculate in your code.)
you refer to https://github.com/IntelRealSense/librealsense/blob/master/src/proc/align.cpp

BTW, don't forget make sure you have enable metadat in your system.
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#update-registry

RealSenseCustomerSupport, 2020-02-11:
----------------------------------------------

@SirNightTW Any update from your side? Thanks!

SirNightTW, 2020-02-11:
Thanks for your reply.
But I'm not familiar with c++, it might be a little bit challenging to me.
I'll update this post if there's any new progress.

The input of align function in python seems to be `pyrealsense2.composite_frame` class
Is there any approach to combine depth and rgb images into `pyrealsense2.composite_frame` ?
If yes, aligning the frames offline might be easier.

RealSenseCustomerSupport, 2020-02-15:
----------------------------------------------

Unfortunately, it's hard to create composite_frame in out of librealsense.
https://github.com/IntelRealSense/librealsense/blob/master/src/archive.cpp#L167

BTW, don't forget to save the intrinsics/extrinsics of these frames.

There is recording/playback mechanism in librealsese. but it may drop some frame according to system load.
https://github.com/IntelRealSense/librealsense/blob/master/src/media/readme.md

RealSenseCustomerSupport, 2020-03-20:
----------------------------------------------

Will close this issue if no more question. Thanks.

closed 2020-03-20 by RealSenseCustomerSupport