realsenseai/librealsense #5090: Unstable depth stream

griffithsleigh, 2019-10-20:
| Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Operating System & Version  |  Linux (Ubuntu 16) | 

### Issue Description
Hi, I use D435i to detect objects. I extract surface normals from depth data. But depth data constantly changes (like fluctuating) even when the camera is stationary. Therefore, normal test scores are always different in my algorithm. Is this normal or is there a way to fix it (get stable depth data)?

agrunnet, 2019-10-20:
Normals are very sensitive to noise. 
Did you try to first post-process by using edge preserving temporal and spatial filters, as per our white paper?

griffithsleigh, 2019-10-21:
Thank you for your reply. Yes, I use temporal and spatial filters. I tried to increase laser power from the controls in the Realsense Viewer app. It got a bit better, but fluctuating continues. Is it possible to enhance depth data using rgb frame data or anything else?

RealSenseCustomerSupport, 2019-10-29:
----------------------------------------------

Using the temporal and spatial filter as agrunnet described is the best way to reduce the noise. Using the high accuracy preset may also help in some cases.
Please let us know if you have any other question. Thank you.

closed 2019-11-05 by RealSenseCustomerSupport