realsenseai/librealsense #5091: Documentation for 'STColorCorrection' (in rs_advanced_mode_command.h)

evercoast-smm, 2019-10-20:
Camera Model: D415
Firmware Version: 05.11.15.00
OS: Ubuntu 18.04 with kernel 4.18.20-041820-generic
Platform: PC
SDK Version: librealsense-2.29.0
Segment: Telecommunications

Question: Please document that `STColorCorrection` matrix that's defined in 'rs_advanced_mode_command.h' and used by the function:

```
void rs2_set_color_correction(rs2_device* dev, const rs2_error** error);
```

where it's used is demonstrated, but not defined, in the *Best-Known-Methods for Tuning Intel® RealSenseTM D400 Depth Cameras for Best Performance* white paper.

Thank you!

MartyG-RealSense, 2019-10-20:
Advanced Mode functions are not fully documented by Intel and such documentation is not planned.  This is because Advanced Mode functions interact with each other in complex ways, and so Intel chose to control them with machine-learning algorithms.  You are free to do trial and error experimentation with the values though to see how changes affect the camera.

evercoast-smm, 2019-10-21:
Thank you, Marty. Having said the above if there's anyone from Intel who could comment on this, it'd be most appreciated. Not asking for color science reference materials, just the nature of the data structure.

All we need are the following:

1) Matrix dimensions, 
2) Does the struct map onto the matrix in column or row major order,
3) And finally, this is a standard CCM? I take it there's nothing unusual here?

Thank you!

RealSenseCustomerSupport, 2019-11-06:
----------------------------------------------

Hi @evercoast-smm,

We have ColorCorrection values showing in the BKM doc and json file. This value is used to remove the impact of IR pattern, only working for D415.
Can we know what you plan to do?

evercoast-smm, 2019-11-06:
Yes, the @agrunnet uses the CCM in the BKM white-paper to remove the IR pattern.

Our application domain is human reconstruction from multiple sensors. We do not run with auto-exposure on. Still, as expected, we note that depending on a variety of factors that the sensor's color profiles do not match.

So, our desire is to do a fairly standard color registration (aka normalization) across all the sensors -- similar to what @agrunnet did to hack the green IR tones away -- but in our case we want all our tones to match. We'd also use a MacBeth chart. We understand how to calculate this, it's fairly standard.

It would save us a lot of time to know the following explicitly from @RealSense-Customer-Engineering:

1) We assume that the data `STColorCorrection` data structure contains a 3x4 matrix. What is its orientation?

2) At what point in image generation process is this matrix applied? E.g. is it on the sensor during debayering, or is it in software libRS after the image has been debayered and transmitted to the host machine?

Thanks! This is really a simple question -- we're not asking for any discussion on color theory.

RealSenseCustomerSupport, 2019-11-07:
----------------------------------------------

Thanks for the explanation. @agrunnet, any comment about this?

agrunnet, 2019-11-07:
I will answer the flow: this is done on-chip. It is in a flow that is parallel to the depth so any changes to color correction do not affect depth at all.

evercoast-smm, 2019-11-07:
Thank you @agrunnet. Your comment raised a question I hadn't previously considered. Is it possible to set the CCM for the color sensor?

RealSenseSupport, 2020-06-29:
Hi @evercoast-smm 

The original question was answered so is there anything further needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

evercoast-smm, 2020-06-30:
Answered. Thank you.

Evercoast <http://www.evercoast.com>  |  Sebastian Marino  |  Co-founder &
CTO

closed 2020-06-30 by RealSenseSupport