realsenseai/librealsense #5092: Error compiling on Raspberry Pi 4

jgwinner, 2019-10-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } although it's N/A | 
| Firmware Version                   | (N/A) | 
| Operating System & Version |   {Raspbian from NOOBS 3_2_0  | 
| Kernel Version (Linux Only)    |  Linux caremon5 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l GNU/Linux   | 
| Platform                                 | Raspberry Pi 4  |
| SDK Version                            |  { just downloaded today }                          | 
| Language                            |  {python3 }                          | 
| Segment			|  {libRealsense build }                          | 

### Issue Description
Linux caremon5 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l GNU/Linux

I did get an error message with this step earlier:

```
root@caremon5:/srv/www/webroot# sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libdrm-amdgpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg

```
<br>

Everything else proceeded OK, except for this step:
<br>
`python3 setup.py build --cpp_implementation`
<br>
```
creating build/temp.linux-armv7l-3.7/google
creating build/temp.linux-armv7l-3.7/google/protobuf
creating build/temp.linux-armv7l-3.7/google/protobuf/pyext
arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I../src -I/usr/include/python3.7m -c google/protobuf/pyext/extension_dict.cc -o build/temp.linux-armv7l-3.7/google/protobuf/pyext/extension_dict.o -Wno-write-strings -Wno-invalid-offsetof -Wno-sign-compare
google/protobuf/pyext/extension_dict.cc: In function ‘PyObject* google::protobuf::python::extension_dict::_FindExtensionByName(google::protobuf::python::ExtensionDict*, PyObject*)’:
google/protobuf/pyext/extension_dict.cc:59:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
google/protobuf/pyext/extension_dict.cc:187:7: note: in expansion of macro ‘PyString_AsStringAndSize’
   if (PyString_AsStringAndSize(arg, &name, &name_size) < 0) {
       ^~~~~~~~~~~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
```

terrorzombie666, 2019-10-22:
same here, following this guide:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

pi@raspberrypi:~ $ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)

pi@raspberrypi:~ $ cmake --version
cmake version 3.13.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).

pi@raspberrypi:~ $ sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libdrm-amdgpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg
```

rohang1997, 2019-11-01:
has someone figured out where this libraries can be downloaded from ? or any link we have add to apt source.list? i am also getting the same error

hannahaponenko, 2019-11-04:
Having the same issue!

suyuan945, 2019-11-15:
I have solved the problem.Here is some change based on origin install command.

compiling on Raspberry Pi 4
$uname -a
::Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
$gcc -v
::gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1) 
$cmake --version
::cmake version 3.13.4

1. Install packages
    you will get an error message for example:Unable to locate package libdrm-amdgpu1-dbg.
    ignore the error and continue the next install command.
2. install protobuf
   change the protobuf version v3.5.1 to v3.10.0
   so change  git clone command for install  the packages :
   `git clone --depth=1 -b v3.10.0 https://github.com/google/protobuf.git`

3. install OpenCV
   I didn't install opencv,this package  seem didn't affect correct open the camera.

4.change pi settings (enable OpenGL)
   insatll pyopengl_accelerate will see an error ,change the command
   $ sudo -H pip3 install pyopengl_accelerate
   to $ sudo -H pip3 install pyopengl_accelerate==3.1.3rc1

Connected D415 to the pi and open terminal
$ realsense-viewer

Good luck
[image]

RealSenseCustomerSupport, 2019-11-22:
----------------------------------------------

@suyuan945, thank you very much for sharing and glad to see D415 working on Pi 4.
Closing this ticket if no further question on the same topic.
Thank you.

closed 2019-11-27 by RealSenseCustomerSupport

jgwinner, 2019-12-22:
When will the official documentation be updated?

rohang1997, 2019-12-22:
The simplest solution
Go to scripts inside repo and download this script-
libuvc_installation.sh
Then 
Chmod +x libuvc_installation.sh
./libuvc_installation
(Will take some time to compile)
Realsense will work for sure

jgwinner, 2019-12-22:
That may be true, but the official docs are really out of date and need to be changed.

Otherwise it's an exercise in frustration for other people to try to make this package work, which is a shame as the Pi 4 is a beast of a machine.

It won't work on a Pi 3 either unless on a very old version of Raspbian.

The docs need to be fixed, and any extra scripts need to be included in the repository.

I realize open source can be really slow to update as it's a DIY mentality, but we paid for this hardware product, and it should run on modern hardware.

beyond21299, 2019-12-30:
> The simplest solution
> Go to scripts inside repo and download this script-
> libuvc_installation.sh
> Then
> Chmod +x libuvc_installation.sh
> ./libuvc_installation
> (Will take some time to compile)
> Realsense will work for sure

I used libuvc_installation.sh to install successfully on TX2 device, but would like to ask how to install pyrealsense2？

prem48science, 2020-09-15:
> same here, following this guide:
> https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md
> 
> ```
> pi@raspberrypi:~ $ uname -a
> Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
> 
> pi@raspberrypi:~ $ gcc -v
> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/8/lto-wrapper
> Target: arm-linux-gnueabihf
> Configured with: ../src/configure -v --with-pkgversion='Raspbian 8.3.0-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-8/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++ --prefix=/usr --with-gcc-major-version-only --program-suffix=-8 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
> Thread model: posix
> gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
> 
> pi@raspberrypi:~ $ cmake --version
> cmake version 3.13.4
> 
> CMake suite maintained and supported by Kitware (kitware.com/cmake).
> 
> pi@raspberrypi:~ $ sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package libdrm-amdgpu1-dbg
> E: Unable to locate package libdrm-exynos1-dbg
> E: Unable to locate package libdrm-freedreno1-dbg
> E: Unable to locate package libdrm-nouveau2-dbg
> E: Unable to locate package libdrm-omap1-dbg
> E: Unable to locate package libdrm-radeon1-dbg
> E: Unable to locate package libdrm-tegra0-dbg
> E: Unable to locate package libdrm2-dbg
> ```
> 
> Edit: Fails the same line
> 
> ```
> creating build/temp.linux-armv7l-3.7
> creating build/temp.linux-armv7l-3.7/google
> creating build/temp.linux-armv7l-3.7/google/protobuf
> creating build/temp.linux-armv7l-3.7/google/protobuf/pyext
> arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I. -I../src -I/usr/include/python3.7m -c google/protobuf/pyext/extension_dict.cc -o build/temp.linux-armv7l-3.7/google/protobuf/pyext/extension_dict.o -Wno-write-strings -Wno-invalid-offsetof -Wno-sign-compare
> google/protobuf/pyext/extension_dict.cc: In function ‘PyObject* google::protobuf::python::extension_dict::_FindExtensionByName(google::protobuf::python::ExtensionDict*, PyObject*)’:
> google/protobuf/pyext/extension_dict.cc:59:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
>         ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
>                       ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
> google/protobuf/pyext/extension_dict.cc:187:7: note: in expansion of macro ‘PyString_AsStringAndSize’
>    if (PyString_AsStringAndSize(arg, &name, &name_size) < 0) {
>        ^~~~~~~~~~~~~~~~~~~~~~~~
> error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1
> ```

I used apt-cache search for libdrm-amdgpu1-dbg it returned libdrm-amdgpu1-dbgsym and libdrm-amdgpu1 so I went on to run command Sudo apt-get install libdrm-amdgpu1-dbgsym which got installed without errors. 

I am trying other steps to install librealsense completely will update if any mistake and solution are found.

scottrpaterson, 2020-10-13:
This issue should not be closed. The docs are still out of date and were never changed.

AyanHBK123, 2020-12-20:
> I have solved the problem.Here is some change based on origin install command.
> 
> compiling on Raspberry Pi 4
> $uname -a
> ::Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
> $gcc -v
> ::gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)
> $cmake --version
> ::cmake version 3.13.4
> 
> 1. Install packages
>    you will get an error message for example:Unable to locate package libdrm-amdgpu1-dbg.
>    ignore the error and continue the next install command.
> 2. install protobuf
>    change the protobuf version v3.5.1 to v3.10.0
>    so change  git  command for install  the packages :
>    `git clone --depth=1 -b v3.10.0 https://github.com/google/protobuf.git`
> 3. install OpenCV
>    I didn't install opencv
> 4. change pi settings (enable OpenGL)
>    install pyopengl_accelerate will see an error ,change the command
>    $ sudo -H pip3 install pyopengl_accelerate
>    to $ sudo -H pip3 install pyopengl_accelerate==3.1.3rc1
> 
> Connected D415 to the pi and open terminal
> $ realsense-viewer
> 
> Good luck
> [image]
copied your all the work and in the end getting this fault 

Segmentation fault

AyanHBK123, 2020-12-20:
> I have solved the problem.Here is some change based on origin install command.
>
> compiling on Raspberry Pi 4
> $uname -a
> ::Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l [...]

I copied all your work but in the end getting this error
when i runned the following command 
realsense-viewer
'Segmentation fault'