realsenseai/librealsense #5095: depth grid

cassie101, 2019-10-21:
while using pyrealsense2, there's a grid pattern coming from depth frame. Is there any way to turn off?

[image]

dorodnic, 2019-10-22:
Hi @cassie101 
What is the SDK version? Camera type?
For the D400 there is really no reason to align depth to IR (they are already aligned)

cassie101, 2019-10-22:
Hi @dorodnic, thanks for your reply 
It was SR300.  if I understand correctly, the depth and IR are also aligned in this model, am I right?
 I was just curious as to why there were grid in the depth display. 
I don’t have my pc with me now, but it was the latest pyrealsense2 that I have via pip

dorodnic, 2019-10-22:
Let me try with the SR300, this may be an artefact of SR300 IR distortion model.

dorodnic, 2019-10-23:
Hi @cassie101 
I see this issue when aligning from depth to IR with the SR300. However, as far as I know, this step can be omitted entirely, since (like with the D400) depth stream is generated from IR stream and they are already aligned.

cassie101, 2019-10-23:
Hi @dorodnic, thank you so much for experiment and quick update. I can understand the situation. Could you please kindly take a look at https://github.com/IntelRealSense/librealsense/issues/5093#issuecomment-544354522 and provide some insight? 
My goal was to align all streams at the best possibility. 
Since depth and IR are aligned by design, could we align (1)depth to rgb and (2) ir to rgb both at the same time? Since when depth-to-rgb alignment is done, rgb image doesn’t have the “artefacts” as shown in https://github.com/IntelRealSense/librealsense/issues/5093#issuecomment-544354522

dorodnic, 2019-10-23:
I see. Yes, this looks like a limitation of the SDK at this point - basically we support alignment of anything to depth and of depth to any stream, but (non-depth)stream to other (non-depth)stream is not supported. Without context, this problem is not solvable (you **need** depth for pixel level alignment), but at the same time passing all tree **should** work. 
Not sure if to label this a bug or an enhancement request, but I struggling to think how this can be currently accomplished, unfortunately. 
Aligning everything **to** depth should work, but in case of SR305 this would involve significant reduction of RGB resolution.

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @cassie101

Is there anything more needed on this ticket?  As outlined by dorodnic there is a limitation here.  We can close this as expected at this point.

closed 2020-01-13 by RealSenseCustomerSupport