realsenseai/librealsense #5098: Unity Wrapper - Get position of hiest or lowest points.

lNeo-afk, 2019-10-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.06.250 | 
| Operating System & Version |   Windows 10  | 
| Platform                                 | PC |
| SDK Version                            |  2.latest }                          | 
| Language                            |  {Unity c# }                          | 

### Issue Description
<I am trying to get the in scene location (x,y,z) for the highest point/points from the depth points. My immediate goal is to assign a simple 3D object (cube) to the highest points and have them move as the as the position of the highest point changes realtime -> ie the Vector3 transform. 

I have tried to modify the Points class to do that but have not been successful at all. I also went through the point cloud scene and code over the past 3 weeks. 

How can I go about doing this? Am I missing something blatant? I would appreciate any leads in the matter. >

lNeo-afk, 2019-10-21:
I was planning on using something like this - which obviously wont work since the shader forces a vertex count. IS there a way to get the position of the points?
`using System.Collections;
using System.Collections.Generic;
using UnityEngine;

public class Zenith : MonoBehaviour
{
    public MeshFilter RTMesh;
    public GameObject cube;

    private Vector3[] verts;
    Vector3 TopVertex;
    // Start is called before the first frame update
    void Start()
    {
    }

    // Update is called once per frame
    void Update()
    {
        verts = RTMesh.sharedMesh.vertices;
        TopVertex = new Vector3(0, float.NegativeInfinity, 0);
        for (int i = 0; i < verts.Length; i++)
        {
            Vector3 vect = transform.TransformPoint(verts[i]);
            if (vect.y > TopVertex.y)
            {
                TopVertex = vect;
                cube.transform.position = TopVertex;
            }
        }
    }
}
`

ogoshen, 2019-11-05:
You should probably work on the frame data directly instead of the mesh.

see [RsPointCloudRenderer.cs](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/unity/Assets/RealSenseSDK2.0/Scripts/RsPointCloudRenderer.cs) for how to get the Points frame and then something like:

```cs
using (var points = fs.FirstOrDefault<Points>(Stream.Depth, Format.Xyz32f))
{
    unsafe {
        Vector3* vertices = (Vector3*)points.VertexData.ToPointer();
        for(int i=0; i<points.Count; ++i) {
            float y = vertices[i].y;
        }
    }
}
```
This avoid copying the data to a managed array.

Now, a note about threads. If this were to execute in the `OnNewSample` callback, it would run on the SDK's thread (which is default, but can be changed in `RsDevice` inspector). Meaning that you won't have access to Unity objects like the `Transform` in `transform.TransformPoint`.
And either way, this computation might take a while, and should not block the frame callback.
So I suggest you use a `FrameQueue`, very much like in `RsPointCloudRenderer`.
You can poll the queue for a frame in `Update`, and do the processing there.

RealSenseCustomerSupport, 2020-01-20:
----------------------------------------------

@lNeo-afk Sorry for the late response. Did you get it through now? Looking forward to your update. Thanks!

closed 2020-04-06 by RealSenseCustomerSupport