realsenseai/librealsense #5099: How to properly use the ROS-bag convert tool

collinabidi, 2019-10-21:
|----------------------------------------|------------------------------------------- |
| Camera Model                        | D400                               | 
| Firmware Version                   |  2.29.0                               | 
| Operating System & Version  |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-31                          | 
| Platform                                  |  PC                                       |
| SDK Version                           |  2.0                                | 
| Language                               |  C++/Python                              | 
| Segment			         |  VR/AR                                     | 
 
### Issue Description
I am attempting to convert the RGB-D .bag file (recorded using the RealSense Viewer tool) to a sequence of RGB/D .png files. However, when I try to cmake the [built-in conversion tool](https://github.com/IntelRealSense/librealsense/tree/master/tools/convert), I get the error:

`
CMake Error at CmakeLists.txt:28 (install):
\tinstall TARGETS given no RUNTIME DESTINATION for executable target "rs-convert"
`

I changed the `${CMAKE_INSTALL_BINDIR}` to a new build file I created, which results in no error, but the files aren't built when I run cmake.

Any tips for building this properly?

MartyG-RealSense, 2019-10-21:
Somebody else who encountered this CMake error when changing the default build location fixed it by adding this line to the top of their cmakelists.txt file to "define some sensible defaults":

**include(GNUInstallDirs)**

https://github.com/leela-zero/leela-zero/issues/1901

ev-mp, 2019-10-22:
@collinabidi hi, all the SDK's "CMaking" must refer to librealsense root directory, as defined in the [installation guide](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#building-librealsense2-sdk). Invoking CMake from sub-directories will end up with missing properties as in the presented case.  
#4747, #4335

Specifically for the `rs-convert` tool the `-DBUILD_EXAMPLES=true` flag should be configured.

collinabidi, 2019-10-22:
Ah, I wasn't aware that the -DBUILD_EXAMPLES would include the tools.

I followed the installation guide and set the flag to true, but the `rs-convert` still doesn't appear in the /usr/local/bin file along with the other `rs-*` files created there.

Is there something else I'm missing?

collinabidi, 2019-10-22:
Ah, never mind, I was able to locate it in the `build` folder. Thank you for the help!

closed 2019-10-22 by collinabidi