realsenseai/librealsense #5100: How to dewrap realtime fisheye video to panoramic/square video in Unity?

johnny-wang16, 2019-10-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { T265 } | 
| Firmware Version                   | (0.1.0.279) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)| 
| SDK Version                            |  { 2.0 }                          | 
| Language                            |  {C#/unity }                          | 
| Segment			|  {VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am using T265 and manged to stream images in realtime in Unity with RsDevice and RsStreamTextureRenderer scripts. I am wondering if it is possible to unwrap the fisheye images to square images in real time.

ogoshen, 2019-10-29:
One way would be to port undistortion code from python [here](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py) to OpenCvSharp\Emgu CV.

Probably a better approach for Unity would be to use shaders to do the same on the GPU.
You'll want to create a 'uvmap', a texture that contains coordinates that map each pixel of the undistorted image to a pixel on the original distorted image.

This is what `initUndistortRectifyMap` from link above does, You could use `rsutil.h`  as a reference:
https://github.com/IntelRealSense/librealsense/blob/81d469db173dd682d3bada9bd7c7570db0f7cf76/include/librealsense2/rsutil.h#L83

ogoshen, 2019-11-03:
Went ahead and created this [gist](https://gist.github.com/ogoshen/37950ac3901080c3716fe56add2fb549)

It's a drop-in replacement for `RsStreamTextureRenderer.cs` and `BW.shader` that undistorts the images.

RealSenseCustomerSupport, 2019-12-24:
----------------------------------------------

@johnny-wang16
There hasn't been any responses since ogoshen added some updates on this thread.  Is there anything else needed here?

johnny-wang16, 2019-12-26:
@ogoshen  thanks for doing this. I ended up with a different approach which is using opencv to undistort the image before passing the datastream into Unity. I'd try your approach if I have time. 

@RealSenseCustomerSupport  I think that's it. thank you @ogoshen  again for creating the replacement.

closed 2019-12-26 by RealSenseCustomerSupport